cocotb/cocotb #3830: `Release` does not work on GHDL

ktbarrett, 2024-04-07:
This affects *all* tests in test_force_release.

```
MODULE=test_force_release TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=vhdl \
 /home/kaleb/.local/bin/ghdl -r  --time-resolution=ps --workdir=sim_build -Psim_build --work=work sample_module  --vpi=/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvpi_ghdl.so  
loading VPI module '/home/kaleb/dev/cocotb/src/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/kaleb/.pyenv/versions/cocotb-system/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     cocotb                             Running on GHDL version 4.0.0-dev (3.0.0.r684.gc84120e00) [Dunoon edition]
     0.00ns INFO     cocotb                             Running tests with cocotb v2.0.0.dev0+5bd689a6 from /home/kaleb/dev/cocotb/src/cocotb
     0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
     0.00ns INFO     cocotb                             Seeding Python random module with 1712516913
     0.00ns INFO     cocotb.regression                  running test_force_release.test_hdl_writes_dont_overwrite_force_combo (1/7)
                                                            Test Forcing then later Releasing a combo signal.
/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/../../../tests/designs/sample_module/sample_module.vhdl:128:7:@0ms:(report note): stream_in_string has been updated, new value is ''
    20.00ns INFO     cocotb.regression                  test_force_release.test_hdl_writes_dont_overwrite_force_combo failed
                                                        Traceback (most recent call last):
                                                          File "/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 30, in test_hdl_writes_dont_overwrite_force_combo
                                                            assert dut.stream_out_data_comb.value == 3
                                                        AssertionError: assert LogicArray('00000101', Range(7, 'downto', 0)) == 3
                                                         +  where LogicArray('00000101', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_out_data_comb).value
                                                         +    where LogicObject(sample_module.stream_out_data_comb) = HierarchyObject(sample_module).stream_out_data_comb
    20.00ns INFO     cocotb.regression                  running test_force_release.test_hdl_writes_dont_overwrite_force_registered (2/7)
                                                            Test Forcing then Releasing a registered output.
    50.00ns INFO     cocotb.regression                  test_force_release.test_hdl_writes_dont_overwrite_force_registered failed
                                                        Traceback (most recent call last):
                                                          File "/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 49, in test_hdl_writes_dont_overwrite_force_registered
                                                            assert dut.stream_out_data_registered.value == 4
                                                        AssertionError: assert LogicArray('00000101', Range(7, 'downto', 0)) == 4
                                                         +  where LogicArray('00000101', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_out_data_registered).value
                                                         +    where LogicObject(sample_module.stream_out_data_registered) = HierarchyObject(sample_module).stream_out_data_registered
    50.00ns INFO     cocotb.regression                  running test_force_release.test_force_followed_by_release_combo (3/7)
                                                            Test if Force followed immediately by Release works on combo signals.
    60.00ns INFO     cocotb.regression                  test_force_release.test_force_followed_by_release_combo failed
                                                        Traceback (most recent call last):
                                                          File "/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 69, in test_force_followed_by_release_combo
                                                            assert dut.stream_out_data_comb.value == 16
                                                        AssertionError: assert LogicArray('00000101', Range(7, 'downto', 0)) == 16
                                                         +  where LogicArray('00000101', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_out_data_comb).value
                                                         +    where LogicObject(sample_module.stream_out_data_comb) = HierarchyObject(sample_module).stream_out_data_comb
    60.00ns INFO     cocotb.regression                  running test_force_release.test_force_followed_by_release_registered (4/7)
                                                            Test if Force followed immediately by Release works on registered signals.
    80.00ns INFO     cocotb.regression                  test_force_release.test_force_followed_by_release_registered failed
                                                        Traceback (most recent call last):
                                                          File "/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 87, in test_force_followed_by_release_registered
                                                            assert dut.stream_out_data_registered.value == 90
                                                        AssertionError: assert LogicArray('00000101', Range(7, 'downto', 0)) == 90
                                                         +  where LogicArray('00000101', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_out_data_registered).value
                                                         +    where LogicObject(sample_module.stream_out_data_registered) = HierarchyObject(sample_module).stream_out_data_registered
    80.00ns INFO     cocotb.regression                  running test_force_release.test_cocotb_writes_dont_overwrite_force_combo (5/7)
                                                            Test Deposits following a Force don't overwrite the value.
   100.00ns INFO     cocotb.regression                  test_force_release.test_cocotb_writes_dont_overwrite_force_combo failed
                                                        Traceback (most recent call last):
                                                          File "/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 110, in test_cocotb_writes_dont_overwrite_force_combo
                                                            assert dut.stream_out_data_comb.value == 10
                                                        AssertionError: assert LogicArray('00001011', Range(7, 'downto', 0)) == 10
                                                         +  where LogicArray('00001011', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_out_data_comb).value
                                                         +    where LogicObject(sample_module.stream_out_data_comb) = HierarchyObject(sample_module).stream_out_data_comb
   100.00ns INFO     cocotb.regression                  running test_force_release.test_cocotb_writes_dont_overwrite_force_registered (6/7)
                                                            Test Deposits following a Force don't overwrite the value.
   140.01ns INFO     cocotb.regression                  test_force_release.test_cocotb_writes_dont_overwrite_force_registered failed
                                                        Traceback (most recent call last):
                                                          File "/home/kaleb/dev/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 143, in test_cocotb_writes_dont_overwrite_force_registered
[... cut ...]
                                                        *******************************************************************************************************************************
                                                        ** TESTS=7 PASS=1 FAIL=6 SKIP=0                                                         160.01           0.01      14667.00  **
                                                        *******************************************************************************************************************************
```

cmarqu, 2024-11-12:
Related: #4150

kms1212, 2024-11-18:
Is there any improvement yet? I have the same problem.

ktbarrett, 2024-11-18:
@kms1212 See #4150. This was fixed recently in GHDL. We aren't closing this issue until #4263 is merged, which requires some unrelated fixes.

PR #4479 (2025-03-04): Improvements to setting values on handles
Closes #4466. Closes #3830.

The highlights are:
* Introduced `get` method for getting values
* `handle.value` getter and setter just use `get()` and `set()`.
* Split the docuemtnation for value properties between `get()` and `set()`.
* Introduced `Immediate` to parallel `Deposit`, `Release`, `Freeze`, and `Force`.
* `setimmediatevalue` was deprecated.
* Fixed bug where `handle.value` setter and `set()` inertially delayed non-inertial actions like Force and Release, whoops...
* Added terms to the glossary related to writing.
* Touched up the documentat [... cut]
closed 2025-03-04 by ktbarrett