cocotb/cocotb #3832: Setting value on forced signal causes implicit release on Riviera

ktbarrett, 2024-04-07:
In `test_force_release.test_cocotb_writes_dont_overwrite_force_registered` when the forced signal is erroneously attempted to be overwritten using a normal deposit, the deposit does not succeed, but it does implicitly `Release` the signal. This causes the process to update the driven signal prematurely causing a test failure.

```
# COUT:    110.00ns INFO     cocotb.regression                  running test_force_release.test_cocotb_writes_dont_overwrite_force_registered (6/7)
# COUT:                                                             Test Deposits following a Force don't overwrite the value.
# COUT:    150.01ns INFO     cocotb.regression                  test_force_release.test_cocotb_writes_dont_overwrite_force_registered failed
# COUT:                                                         Traceback (most recent call last):
# COUT:                                                           File "/opt/actions-runner/_work/cocotb/cocotb/tests/test_cases/test_force_release/test_force_release.py", line 116, in test_cocotb_writes_dont_overwrite_force_registered
# COUT:                                                             assert dut.stream_out_data_registered.value == 10
# COUT:                                                         AssertionError: assert LogicArray('01001101', Range(7, 'downto', 0)) == 10
# COUT:                                                          +  where LogicArray('01001101', Range(7, 'downto', 0)) = LogicObject(sample_module.stream_out_data_registered).value
# COUT:                                                          +    where LogicObject(sample_module.stream_out_data_registered) = HierarchyObject(sample_module).stream_out_data_registered
```

imphil, 2024-06-11:
Filed [SPT82777](https://www.aldec.com/en/support/customer_portal/cases/22777) with Aldec.

imphil, 2025-07-10:
Fixed in Riviera-PRO 2024.10, according to the changelog:
> An issue with forcing a signal with two vpi_put_value() subroutine calls among which one was invoked with the vpiForceFlag flag and the other with the vpiInertialDelay flag is resolved. (SPT82777)

closed 2025-07-10 by imphil