cocotb/cocotb #3846: COCOTB_RESOLVE_X not working

JerryWuZiJie, 2024-04-17:
I've experience the issue that setting the `COCOTB_RESOLVE_X` flag in makefile is not working. It only work if I export it as environment variables

cocotb version: COCOTB_RESOLVE_X
operating system: Red Hat Enterprise Linux Server ver 7.9
simulator: xcelium 19.03-s010
Python: 3.6.8 system version

ktbarrett, 2024-04-17:
Please post exact code.

I'm guessing you are doing something like this in the makefiles.
```make
COCOTB_RESOLVE_X = whatever
```
This does not export the make variable as an environment variable. you need to add the `export` keyword.
```make
export COOTB_RESOLVE_X = whatever
```

The only reason this is not needed for things like `TESTCASE`, `MODULE`, etc. is because the makefiles export those for you. But they don't export *all* environment variables.

Also, `COCOTB_RESOLVE_X` and `BinaryValue` were removed in 2.0, so expect casting values coming from handles to `int` to always act like `VALUE_ERROR` from now on.

JerryWuZiJie, 2024-04-17:
I see thanks. I thought all the variables listed [here](https://docs.cocotb.org/en/stable/building.html) are makefile vars. Apparently I missed some lines. Thanks for pointing it out!

closed 2024-04-17 by JerryWuZiJie

ktbarrett, 2024-04-17:
Sorry, the documentation is not clear about which variables are just for Make vs environment variables passed to Python, and which environment variables are exported for you.