cocotb/cocotb #3850: Python Runner starts tests directly (no waiting for GUI)

AmutioAlex, 2024-04-19:
I have been using cocotb for a while and decided to switch to the Python runner approach instead of using a Makefile.

The problem that I found is that, when I use Xcelium as a simulator and open the GUI, the test starts right away and does not wait to be started from the GUI. This is the current code that I run directly, **not with pytest**:

```python
runner = get_runner("xcelium")
runner.build(verilog_sources    = [...],
             hdl_toplevel       = ...,
             always             = True)
runner.test(hdl_toplevel_lang   = "verilog",
            hdl_toplevel        = ...,
            test_module         = ...,
            gui                 = True)
```
Note that the GUI is opened, thus the _gui_ flag works correctly. The problem is that before the GUI loads the test has already started. It is not being triggered by the user.

I took a look at the source code of [runner.py](https://github.com/cocotb/cocotb/blob/stable/1.8/cocotb/runner.py) for the version that I am using (cocotb 1.8.1) and I am guessing that the main culprit here are lines [1002](https://github.com/cocotb/cocotb/blob/stable/1.8/cocotb/runner.py#L1002) and [1003](https://github.com/cocotb/cocotb/blob/stable/1.8/cocotb/runner.py#L1003), which always include the following options to the `xrun` command: `-input @run` and `-input @exit`.

I first thought that this was being fixed already for the next release, but the problem persists on the [runner.py](https://github.com/cocotb/cocotb/blob/master/src/cocotb_tools/runner.py) file of the `master` branch. Lines [1352](https://github.com/cocotb/cocotb/blob/master/src/cocotb_tools/runner.py#L1352) to [1360](https://github.com/cocotb/cocotb/blob/master/src/cocotb_tools/runner.py#L1360) state that in every case the test is run right away.

I think it would be wise for the runner to behave like the Makefile in this sense: if the GUI is started then the user has to trigger it.

Further information of my setup in case it is needed:
- cocotb 1.8.1
- Red Hat Enterprise 7.9 64-bit
- Xcelium 21.09.s003
- Python 3.10.1

ktbarrett, 2024-04-19:
Sounds appropriate to me, feel free to provide a PR.

s3bs, 2024-07-03:
IMHO this is a problem with the Makefile and not with the runner. 
Changing GUI or WAVES between runs will not trigger a regeneration of the `runsim.do` but it should. 
https://github.com/cocotb/cocotb/blob/ca6a97ff4d6af2ac13d0271cb4f8ed0b01e27627/src/cocotb_tools/makefiles/simulators/Makefile.questa#L143-L152

imphil, 2024-07-08:
Not ready for 1.9, dropping milestone.