cocotb/cocotb #3858: `TypeError` exception when running simulation

cjmeyer, 2024-04-22:
When running the example `matrix_multiplier` from the `stable/1.8` branch, I get a `TypeError` exception being thrown from `subprocess.py` as a result of calling `os.fsencode`. I was able to track it down a bit, and it seems that the environment variables that are being passed to the subprocess from `runner.py:359` include an environment variable with an empty value, namely `LIBPYTHON_LOC`. I found [here](https://docs.cocotb.org/en/stable/custom_flows.html) that the value for this environment variable is supplied from `find_libpython`. If I run `find_libpython` from the console, it returns an empty string.

cocotb version - 1.8, stable
operating system - Ubuntu 20.04.6 LTS, 64-bit
simulator version - GHDL, 4.0.0
python version - 3.12, in a virtual environment
error:

```
/media/storage/<USER>/cocotb/examples/matrix_multiplier/tests/./test_matrix_multiplier.py:16: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  from cocotb.runner import get_runner
INFO: Running command ghdl -i --work=top --std=08 /media/storage/<USER>/cocotb/examples/matrix_multiplier/hdl/matrix_multiplier_pkg.vhd /media/storage/<USER>/cocotb/examples/matrix_multiplier/hdl/matrix_multiplier.vhd in directory /media/storage/<USER>/cocotb/examples/matrix_multiplier/tests/sim_build
INFO: Running command ghdl -m --work=top --std=08 matrix_multiplier in directory /media/storage/<USER>/cocotb/examples/matrix_multiplier/tests/sim_build
analyze /media/storage/<USER>/cocotb/examples/matrix_multiplier/hdl/matrix_multiplier_pkg.vhd
analyze /media/storage/<USER>/cocotb/examples/matrix_multiplier/hdl/matrix_multiplier.vhd
elaborate matrix_multiplier
INFO: Running command ghdl -r --work=top --std=08 matrix_multiplier --vpi=/media/storage/<USER>/fpga-project/.venv/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_ghdl.so -gDATA_WIDTH=32 -gA_ROWS=10 -gB_COLUMNS=4 -gA_COLUMNS_B_ROWS=6 in directory /media/storage/<USER>/cocotb/examples/matrix_multiplier/tests/sim_build
Traceback (most recent call last):
  File "/media/storage/<USER>/cocotb/examples/matrix_multiplier/tests/./test_matrix_multiplier.py", line 284, in <module>
    test_matrix_multiplier_runner()
  File "/media/storage/<USER>/cocotb/examples/matrix_multiplier/tests/./test_matrix_multiplier.py", line 275, in test_matrix_multiplier_runner
    runner.test(
  File "/media/storage/<USER>/fpga-project/.venv/lib/python3.12/site-packages/cocotb/runner.py", line 320, in test
    self._execute(cmds, cwd=self.test_dir)
  File "/media/storage/<USER>/fpga-project/.venv/lib/python3.12/site-packages/cocotb/runner.py", line 359, in _execute
    process = subprocess.run(cmd, cwd=cwd, env=self.env)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.12/subprocess.py", line 1870, in _execute_child
    env_list.append(k + b'=' + os.fsencode(v))
                               ^^^^^^^^^^^^^^
  File "<frozen os>", line 818, in fsencode
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

ktbarrett, 2024-04-22:
`find_libpython` should not return nothing. I'm not sure where you got your Python environment from, but does it include a libpython.so? This is required.

cjmeyer, 2024-04-22:
Python 3.12 was installed from `jammy-updates` using `apt install python3.12`. I searched for `libpython3.12.so` and also was not able to find it. I ran `apt install python3.12-dev` and then I was able to find `libpython3.12.so`, however, the installation instructions for cocotb at https://docs.cocotb.org/en/stable/install.html, for Linux - Debian, do not list `libpython3-dev` as needing to be installed. So, my current questions are...

1. Should the installation instructions for cocotb include needing to install `libpython` for Linux - Debian systems, that is should the installation command be `apt install python3 python3-dev python3-pip` (There is actually no `python3.12-pip` package to be installed in my case).

2. For robustness, it may be nice if cocotb checked the return value from `find_libpython.find_libpython()` for a `None` or empty string return and raised a more informative error about not being able to find the appropriate shared python library.

imphil, 2024-04-22:
Great you figured this out, @cjmeyer! 

On 1) libpython.so shouldn't be part of the dev package. But I couldn't double check: I couldn't find python3.12 in jammy-updates. But a quick look indicates that there is libpython*-minimal and libpython* in Ubuntu (probably recently?) The minimal package does not seem to include the shared library? It would be great if you could do a bit more digging here.

On 2) Would you mind opening a pull request to improve the situation?

ktbarrett, 2024-04-22:
Yeah the installation instructions used to list `python3-dev` which pulled in `libpython3-dev` on distros that separated the two. When we moved to providing wheels we removed that requirement, which lost the implicitly dependency on libpython on Debian/Ubuntu.

The later is because we "fallback" to attempting to load `libpython3.so` which finds the version of libpython for 3.8 or whatever is standard on 20.04 that you also have installed, leading to a misconfigured environment and a failure to encodings (tell tale sign of misconfigured Python environment).

cjmeyer, 2024-04-22:
@imphil - I would love to open a pull request to add the check. To that end...

@ktbarrett - I am not sure I understand about the "fallback" to libpython3.so. Looking at [runner.py:136](https://github.com/cocotb/cocotb/blob/248b95c5ad4074539ece038fc07632b5f634708a/src/cocotb_tools/runner.py#L136), it seems that the `LIBPYTHON_LOC` environment variable is either set or not set. Is there fallback logic located somewhere else?

cjmeyer, 2024-04-22:
@imphil - My ability to check the different environments might be a little limited as I am not the administrator of the system that this is installed on. The searching I did via the online Ubuntu repositories seem to indicate that `libpython3.12` is not installed as part of a `default` installation. That said, I am working on an older Ubuntu distribution so the situation may be different on newer versions of Ubuntu. What I am unsure of is if this applies to more Debian type systems than just Ubuntu. Not sure how to answer that question.

ktbarrett, 2024-04-22:
It's built into embed.cpp by cocotb_build_libs.py. Definitely not the most obvious or foolproof, it should definitely be done away with, so proper checking in the runner and Makefiles would be a good step towards that goal.

ktbarrett, 2024-04-23:
I'm okay with that.

cjmeyer, 2024-04-29:
@imphil, I submitted a pull request, #3868. I didn't indicate that it would close this issue since I wasn't sure if any additional documentation or installation instructions still needed updating.

ktbarrett, 2024-05-03:
> * There is no unversioned libpython3 package

It looks like `libpython3-dev` in Debian and Ubuntu and `python3-libs` in Red Hat distros would work?

PR #3879 (2024-05-05): Add libpython package to normal installation instructions
Closes #3858. Updates the installation docs to recommend users install packages which include `libpython3.so`.
closed 2024-05-05 by ktbarrett