cocotb/cocotb #3894: SIM=Verilator, WAVES=1, COMPILE_ARGS+=--trace no longer produces dump.vcd

ian-l-kennedy, 2024-05-14:
@toddstrader it seems that WAVES=1 and COMPILE_ARGS += --trace no longer produces a dump.vcd when using SIM=verilator

62f1839525e56ee6fe17f558d49aa513fbc7e745 it works
 
f7b37ca50ccd72ba99a608f5d0acc634523bfcab it broke (@toddstrader this is your commit)

Please advise!

toddstrader, 2024-05-14:
Ah, yup.  So, `--trace` is not a verilation-time parameter, it is something that you need to pass to the verilated binary.  From looking at `Makefile.verilator` I think you could `PLUSARGS += --trace` instead.  Can you try that?  Also, I don't think the Verilator Makefile cares about `WAVES`.  But if my suggestion works, then maybe we could use it to supply `--trace`?

ian-l-kennedy, 2024-05-14:
Hi, thanks for getting back to me!
I will give that a try tomorrow and get back to ya with my findings.
I am not sure about your assertion about verilation time, and the only reason I say that is because there is standard out text from the verilator steps before the invocation of the executable (the sim) that includes text along the lines of TRACE=1, FST=0, and VCD=1 (and all those values switch as I adjust the COMPILE_ARGS). I will post the exact text that I am discussing tomorrow when I am back at my desk. Please excuse my ignorance, I do not have any knowledge as a contributor, but only as a user.

toddstrader, 2024-05-15:
Whoops.  You're right.  Well, we're both right.  `--trace` is both a verilation time parameter:
```
verilator --help | grep "\<trace\> "
        --trace                     Enable waveform creation
```
and a run time parameter (for binaries built by cocotb):
```
tests/test_cases/test_defaultless_parameter/sim_build/Vtop --help |& grep -- "--trace "
  --trace      Enables tracing (VCD or FST)
```

The former bakes resources into the verilated C++ to be able to produce waves, whereas the latter says that you actually want the waves for this run.  And yeah, the runtime flag was added in that commit you mentioned in order to be more compatible with the python runner.

I thought we could just bury the new run time flag under `VERILATOR_TRACE` but apparently that has been deprecated:
```
/usr/scratch/local/devs/cocotb-stock/src/cocotb_tools/makefiles/Makefile.deprecations:6: VERILATOR_TRACE is deprecated, see the "Simulator Support" section in the documentation.
```

This is how the docs currently say to produce waves with Verilator:
https://docs.cocotb.org/en/stable/simulator_support.html#sim-verilator-waveforms

which is now not entirely true.  This works for me now:
```
make -C tests/test_cases/test_defaultless_parameter SIM=verilator EXTRA_ARGS+="--trace --trace-structs" PLUSARGS="--trace"
```
but is a bit verbose.   Can you give that a shot and see if it works?

Beyond that, @ktbarrett / @imphil should we just add `PLUSARGS = --trace` to the docs?  Should we add a new variables (maybe `VTOP_ARGS`) to make it more clear what is happening here?  And should we wrap this all up in one Makefile switch so as to make things easier?  It seems like that was the intent of `VERILATOR_TRACE` but that is deprecated now, so that feels like going backwards.

dpetrisko, 2024-05-16:
Adding that "PLUSARGS = --trace --trace-file foobar.vcd" will rename the .vcd would be a useful addition to the doc

paul-demo, 2024-07-25:
I'm using the cocotb test runner, and upgrading from cocotb v1.8.1 to v1.9 waves disappeared for me too.

I now include these in my build args (they maybe are duplicated by cocotb's runner):
```
"--trace",
"--trace-fst",
"--trace-structs",
```

And also make sure to include `waves=True` when invoking both `build()` and `test()` for the runner.

This restores waves.

ktbarrett, 2024-07-25:
Seems like some work needs to be done so that `waves` is passed to `build` and supplies those build arguments implicitly. We are open to PRs.

WernerFS, 2024-10-16:
> I'm using the cocotb test runner with verilator 5.024, and upgrading from cocotb v1.8.1 to v1.9 waves disappeared for me too.
> 
> I now include these in my build args (they maybe are duplicated by cocotb's runner):
> 
> ```
> "--trace",
> "--trace-fst",
> "--trace-structs",
> ```
> 
> And also make sure to include `waves=True` when invoking both `build()` and `test()` for the runner.
> 
> This restores waves.

I tested your solution on cocotb==2.0.0.dev0+fa3e92c1 using the "test_dff.py" without any luck. VCD file won't be generated although waves is set to True.

I checked with the Makefile approach and it works there. Using the EXTRA_ARGS += --trace. Not sure what is going wrong when execute_cmds is called. I checked how the cmds when calling from the Makefile differ from the one built by the runner.

There is a couple differences:
Makefile:

`
/home/werner/oss-cad-suite/bin/verilator -cc --exe -Mdir sim_build --top-module dff --vpi --public-flat-rw --prefix Vtop -o dff -LDFLAGS -Wl-rpath/home/werner/oss-cad-suite/lib/python3.11/site-packages/cocotb-2.0.0.dev0+fa3e92c1-py3.11-linux-x64.egg/cocotb/libs -L/home/werner/oss-cad-suite/lib/python3.11/site-packages/cocotb-2.0.0.dev0+fa3e92c1-py3.11-linux-x64.egg/cocotb/libs -lcocotbvpi_verilator --trace /cocotb/examples/simple_dff/dff.sv /home/werner/oss-cad-suite/lib/python3.11/site-packages/cocotb-2.0.0.dev0+fa3e92c1-py3.11-linux-x64.egg/cocotb/share/lib/verilator/verilator.cpp
`

runner (output was edited to show relevant changes)

`
/home/werner/oss-cad-suite/bin/verilator -cc --exe -Mdir sim_build --top-module dff --vpi --public-flat-rw --prefix Vtop -o dff -LDFLAGS -Wl-rpath/home/werner/oss-cad-suite/lib/python3.11/site-packages/cocotb-2.0.0.dev0+fa3e92c1-py3.11-linux-x64.egg/cocotb/libs -L/home/werner/oss-cad-suite/lib/python3.11/site-packages/cocotb-2.0.0.dev0+fa3e92c1-py3.11-linux-x64.egg/cocotb/libs -lcocotbvpi_verilator --trace /cocotb/examples/simple_dff/dff.sv /home/werner/oss-cad-suite/lib/python3.11/site-packages/cocotb-2.0.0.dev0+fa3e92c1-py3.11-linux-x64.egg/cocotb/share/lib/verilator/verilator.cpp
`

So in practice, the name of the output executable is different. I also noticed that when running the Make file it calls the simulation on Vtop but the runner renames the executable to dff. Not sure if this may be the issue. I will continue to investigate. 

Let me know if there is anything else I can try.

WernerFS, 2024-10-19:
You are right, the issue was somewhere else. #2238 solves this issue taking care of the EXTRA_ARGS directly in `Makefile.verilator`.