cocotb/cocotb #3904: Unable to Read Full 4959-bit Register

MrWater98, 2024-05-21:
#### Description
I defined a 4959-bit wide register named `coverage` in my Verilog code. However, when I try to read this register using Cocotb, I can only retrieve 2048 bits. Here are the relevant code snippets and debug information.

#### Verilog Code
```verilog
wire [31:0] wb_pc;
reg [4959:0] coverage = 4960'b0;
```

#### Cocotb Test Code
coverage_ = getattr(dut, "coverage_").value
str_cov = str(coverage_)[::-1]
_cov_path.append(str_cov)

#### Debug Information

In the Cocotb debug console, the length of the register is correctly shown as 4959 bits:
len: 4959

But when reading the value using Cocotb, only 2048 bits are retrieved:
len(getattr(dut, "coverage_").value.binstr) = 2048

Screenshots

[image]

Expected Behavior
Cocotb should be able to read the full 4959-bit register.

Actual Behavior
Cocotb only reads 2048 bits of the 4959-bit register.

Environment
Cocotb version: 1.8.1
Operating System: Ubuntu 20.04.6 LTS on Windows 10 x86_64
Kernel: 5.15.123.1-microsoft-standard-WSL2
Simulator: Verilator

MrWater98, 2024-05-21:
I found that this issue does not occur when switching the simulator to Icarus Verilog.

ktbarrett, 2024-05-21:
Cocotb does not put limits on the size of objects, we simply return whatever the VPI gives us. This is probably an upstream issue, try opening an issue against Verilator.

ktbarrett, 2024-05-27:
Upstream issue reported.

closed 2024-05-27 by ktbarrett