cocotb/cocotb #3909: When I simulate normally in windows environment, but change to linux environment reported such an error, what is the solution

jj319410, 2024-05-29:
-.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 10.2 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /usr/local/lib64/python3.6/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1716975169
     0.00ns CRITICAL cocotb.regression                  Failed to import module test_axis_fifo_adapter: No module named 'cocotb._vendor.find_libpython'
     0.00ns INFO     cocotb.regression                  MODULE variable was "test_axis_fifo_adapter"
     0.00ns INFO     cocotb.regression                  Traceback: 
     0.00ns INFO     cocotb.regression                  Traceback (most recent call last):
                                                          File "/usr/local/lib64/python3.6/site-packages/cocotb/regression.py", line 203, in _discover_tests
                                                            module = _my_import(module_name)
                                                          File "/usr/local/lib64/python3.6/site-packages/cocotb/regression.py", line 72, in _my_import
                                                            mod = __import__(name)
                                                          File "/usr/local/lib/python3.6/site-packages/_pytest/assertion/rewrite.py", line 171, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/usr/project/AXIS-ETHERNET/lib/axis/tb/axis_fifo_adapter/test_axis_fifo_adapter.py", line 31, in <module>
                                                            import cocotb_test.simulator
                                                          File "/usr/local/lib/python3.6/site-packages/_pytest/assertion/rewrite.py", line 171, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/usr/local/lib/python3.6/site-packages/cocotb_test/simulator.py", line 13, in <module>
                                                            import cocotb._vendor.find_libpython as find_libpython
                                                        ModuleNotFoundError: No module named 'cocotb._vendor.find_libpython'

jj319410, 2024-05-29:
[image]

ktbarrett, 2024-05-29:
Looks like you need to update cocotb_test. We don't vendor in find_libpython anymore.

jj319410, 2024-05-30:
[image]
the cocotb_test should be updated?

ktbarrett, 2024-05-30:
Yes. `pip install --upgrade cocotb_test`. I see in your previous comment in the traceback that you are importing `cocotb_test` and it is trying to import `cocotb._vendor.find_libpython` which doesn't exist in cocotb 1.8. Presumably, `cocotb_test` has been updated to do the right thing. If upgrading doesn't fix it, file a bug reporter with `cocotb_test` upstream.

closed 2024-07-07 by github-actions[bot]