cocotb/cocotb #3912: Verilator Compile Error

Devil-SX, 2024-06-02:
Environments
```
operating system WSL Ubuntu 20.04
cocotb version 1.8.1 (install from pip)
verilator version 5.023 devel rev v5.020-176-g953249aa4 (build from source)
python version 3.10.14 (in conda environment)
```

Error
```
g++    verilator.o verilated.o verilated_dpi.o verilated_vpi.o verilated_threads.o Vtop__ALL.a   -Wl,-rpath,/home/saver/tools/miniconda3/envs/torch2_2/lib/python3.10/site-packages/cocotb/libs -L/home/saver/tools/miniconda3/envs/torch2_2/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator  -pthread -lpthread -latomic   
/usr/bin/ld: verilator.o: in function `main':
verilator.cpp:(.text.startup+0x4c): undefined reference to `Vtop::Vtop(char const*)'
/usr/bin/ld: verilator.cpp:(.text.startup+0x8b): undefined reference to `Vtop::eval_step()'
/usr/bin/ld: verilator.cpp:(.text.startup+0xc5): undefined reference to `Vtop::eventsPending()'
/usr/bin/ld: verilator.cpp:(.text.startup+0xd9): undefined reference to `Vtop::nextTimeSlot()'
/usr/bin/ld: verilator.cpp:(.text.startup+0x115): undefined reference to `Vtop::final()'
collect2: error: ld returned 1 exit status
make[2]: *** [Vtop.mk:66: Vtop] Error 1
make[2]: Leaving directory '/home/saver/workspaces/Chimera/scan_chisel/sim_build'
make[1]: *** [/home/saver/tools/miniconda3/envs/torch2_2/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:61: sim_build/Vtop] Error 2
make[1]: Leaving directory '/home/saver/workspaces/Chimera/scan_chisel'
make: *** [/home/saver/tools/miniconda3/envs/torch2_2/lib/python3.10/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

My makefile and python file is 
```makefile
# Default simulator
SIM ?= verilator
TOPLEVEL_LANG ?= verilog

# Verilog sources
VERILOG_SOURCES += $(PWD)/verilog/MacReduce.sv
# Use VHDL_SOURCES for VHDL files

# Top-level module name
TOPLEVEL = MacReduce

# Python test module name
MODULE = get_started

# Set PYTHONHOME to the found Python path
export PYTHONHOME := $(shell which python)

# Include cocotb's make rules for simulator setup
include $(shell cocotb-config --makefiles)/Makefile.sim
```

get_started.py
```python
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
async def my_first_test(dut):
    """Try accessing the design."""

    for cycle in range(10):
        dut.clock.value = 0
        await Timer(1, units="ns")
        dut.clock.value = 1
        await Timer(1, units="ns")

    dut._log.info("my_signal_1 is %s", dut.io_a_0.value)
    # assert dut.my_signal_2.value[0] == 0, "my_signal_2[0] is not 0!"

```

ktbarrett, 2024-06-03:
I'm not entirely sure here. Have you tried using a released version of Verilator? v5.024 works for me (HEAD is currently broken because of changes in Verilator). Make sure you `make clean` or otherwise clean up the built design before trying again; the Verilator makefile doesn't rebuild on every call so you might be accidentally using stale broken compiles. I've been there and done that.

Devil-SX, 2024-06-23:
I've solved the problem. The problem was caused by cocotb not recognising the conda virtual environment.

`cocotb/share/makefiles/Makefile.inc` 

```diff
- IS_VENV=$(shell $(shell cocotb-config --python-bin) -c 'import sys; print(sys.prefix != sys.base_prefix)')
+ IS_VENV=$(shell $(shell cocotb-config --python-bin) -c 'import sys, os; print((sys.prefix != sys.base_prefix) or ("CONDA_PREFIX" in os.environ))')
```

ktbarrett, 2024-06-23:
Ah. Fun. Conda gets to be special. I think the ultimate solution is to stop setting PYTHONHOME as I mentioned [here](https://github.com/cocotb/cocotb/issues/3643#issuecomment-1879758728), but a PR with that fix would be welcome.