cocotb/cocotb #3918: ValueError: '/usr/local/bin/cocotb-clean' does not start with '/usr/local/lib/python3.7/site-packages'

jj319410, 2024-06-04:
Launching pytest with arguments /usr/project/AXIS-ETHERNET/lib/axis/tb/axis_fifo_adapter/test_axis_fifo_adapter.py --no-header --no-summary -q in /usr/project/AXIS-ETHERNET/lib/axis/tb/axis_fifo_adapter

Traceback (most recent call last):
  File "/usr/project/pycharm-community-2024.1.1/plugins/python-ce/helpers/pycharm/_jb_pytest_runner.py", line 60, in <module>
    sys.exit(pytest.main(args, plugins_to_load + [Plugin]))
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 150, in main
    config = _prepareconfig(args, plugins)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 332, in _prepareconfig
    pluginmanager=pluginmanager, args=args
  File "/usr/local/lib/python3.7/site-packages/pluggy/_hooks.py", line 433, in __call__
    return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
  File "/usr/local/lib/python3.7/site-packages/pluggy/_manager.py", line 112, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.7/site-packages/pluggy/_callers.py", line 133, in _multicall
    teardown[0].send(outcome)
  File "/usr/local/lib/python3.7/site-packages/_pytest/helpconfig.py", line 104, in pytest_cmdline_parse
    config: Config = outcome.get_result()
  File "/usr/local/lib/python3.7/site-packages/pluggy/_result.py", line 108, in get_result
    raise exc.with_traceback(exc.__traceback__)
  File "/usr/local/lib/python3.7/site-packages/pluggy/_callers.py", line 80, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 1075, in pytest_cmdline_parse
    self.parse(args)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 1425, in parse
    self._preparse(args, addopts=addopts)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 1300, in _preparse
    self._consider_importhook(args)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 1204, in _consider_importhook
    self._mark_plugins_for_rewrite(hook)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 1224, in _mark_plugins_for_rewrite
    for name in _iter_rewritable_modules(package_files):
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 863, in _iter_rewritable_modules
    package_files = list(package_files)
  File "/usr/local/lib/python3.7/site-packages/_pytest/config/__init__.py", line 1221, in <genexpr>
    for file in dist.files or []
  File "/usr/local/lib/python3.7/site-packages/importlib_metadata/__init__.py", line 515, in files
    or self._read_files_egginfo_sources()
  File "/usr/local/lib/python3.7/site-packages/importlib_metadata/_functools.py", line 102, in wrapper
    return func(param, *args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/importlib_metadata/__init__.py", line 509, in skip_missing_files
    return list(filter(lambda path: path.locate().exists(), package_paths))
  File "/usr/local/lib/python3.7/site-packages/importlib_metadata/__init__.py", line 550, in <genexpr>
    for name in text.splitlines()
  File "/usr/local/lib/python3.7/pathlib.py", line 900, in relative_to
    .format(str(self), str(formatted)))
ValueError: '/usr/local/bin/cocotb-clean' does not start with '/usr/local/lib/python3.7/site-packages'

ktbarrett, 2024-06-04:
This looks to be an issue with cocotb-test, or maybe even pytest. Are you running an up to date pytest and cocotb-test? Either way, please report issues in the appropriate repo. https://github.com/themperek/cocotb-test.

closed 2024-06-04 by ktbarrett

jj319410, 2024-06-04:
[image]