cocotb/cocotb #3935: VCS (and ActiveHDL) simulator are not supported in latest stable or latest dev

rohitece06, 2024-06-11:
Documentation says that vcs is supported https://docs.cocotb.org/en/latest/simulator_support.html#synopsys-vcs

but I get error when trying to use vcs

Cocotb - v1.8.1, v2.0.0.dev0, 64 bit,
Rocky8 (CentOS, 64bit) 
Python 3.12.4 in Conda on linux

Error output from 2.0.0.dev0 

   File "/home/redacted/miniconda3/lib/python3.12/site-packages/cocotb_tools/runner.py", line 1431, in get_runner
    raise ValueError(
ValueError: Simulator 'vcs' is not in supported list: icarus, questa, ghdl, riviera, verilator, xcelium, nvc

imphil, 2024-06-20:
That being said, it shouldn't be overly hard to add support for VCS or ActiveHDL to the runner by just copying over one existing class and modifying it with the arguments passed in the Makefile for the respective simulator. Would you mind doing a PR for that once you got it working, @rohitece06?

rohitece06, 2024-06-20:
I was able to get it to work (at least invoke vcs) with Makefile. Running vcs on the remote has different problems specific to my environment. 

I will try to find some bandwidth to include vcs in the python runner.

closed 2024-06-20 by rohitece06

lihq0716, 2024-08-05:
Hey, how can I use vcs for simulation