cocotb/cocotb #3947: Renamed variable COCOTB_COVERAGE_RCFILE won't have an effect

cmarqu, 2024-06-18:
I think the PR that did the renaming of the COVERAGE_RCFILE variable in the code has slightly broken the coverage functionality.
Unless I'm misunderstanding things, we did rely on `coverage.py` to read the original name `COVERAGE_RCFILE` (see https://coverage.readthedocs.io/en/latest/config.html#configuration-reference), which it won't do for `COCOTB_COVERAGE_RCFILE`.

_Originally posted by @cmarqu in https://github.com/cocotb/cocotb/pull/3943#discussion_r1645031484_

cmarqu, 2024-06-18:
Re-exporting this variable under its original name would help - or we revert the name change.
We should have a comment that explains the fact that it's `coverage.py` which actually does interpret the variable.
We should also have a test for this functionality.

teobiton, 2024-06-18:
I investigated a bit and there is also the possibility to pass the `config_filepath` as an argument of `coverage.coverage()`, and it will use that. The coverage module only looks at `COVERAGE_RCFILE` if `config_path`  argument of the Coverage class is left to its default value.

ktbarrett, 2024-06-18:
> I investigated a bit and there is also the possibility to pass the config_filepath as an argument of coverage.coverage(), and it will use that.

I like this option. That should be a simple change, right?

PR #3950 (2024-06-19): Fix config_file for coverage
The configuration file is given to the coverage initializer via an argument instead of it being read from the environment in the coverage module.

Fixes #3947
closed 2024-06-19 by ktbarrett