cocotb/cocotb #3955: Xcelium python runner fails to generate waves

Jonah-Foley, 2024-06-20:
## Problem Statement
Coco.tb version: 1.8.1
Operating system: Rocky Linux 8 (Equivalent to RHEL8)
Simulator: Xcelium 2403
Python Version: Python 3.11 local install

When running a simulation via the python runner with the argument `waves=True` to `runner.test()` the following error is seen:

```
xrun: *E,OPT2ND: option (-input) requires an argument.
xrun: *E,BDOPTQ: Unknown option "-input "@database -open cocotb_waves -default" -input "@probe -database cocotb_waves -create boxcar -all -depth all" -input "@run" -input "@exit" ".
F
```
## Proposed solution

Swap 
```
+ ["-input"]
+ [
    f'-input "@database -open cocotb_waves -default" '
    f'-input "@probe -database cocotb_waves -create {xrun_top} -all -depth all" '
    f'-input "@run" '
    f'-input "@exit" '
    if self.waves
    else "@run; exit;"
]
```
for
```
+ ["-input"]
+ [
      f"@database -open cocotb_waves -default;"
      f"probe -database cocotb_waves -create {xrun_top} -all -depth all;"
      f"run;"
      f"exit;"
      if self.waves
      else "@run; exit;"
  ]
```
in `Xcelium._test_command()` of `cocotb.runner.py`

ktbarrett, 2024-06-20:
@Jonah-Foley We are open to pull requests. Seems like a simple one.

Jonah-Foley, 2024-06-20:
@ktbarrett on it.

Jonah-Foley, 2024-06-20:
@ktbarrett This does not seem to be an issue on newer versions of coco.tb (master works fine) - would you still like a PR that targets v1.8 (what im using)?

ktbarrett, 2024-06-20:
@imphil Is #3447 slated to be included in the backports for 1.9?

@Jonah-Foley You probably don't need to do a PR. We'll make sure that #3447 ends up in the 1.9 release, which should be releasing "soon".

Jonah-Foley, 2024-06-20:
Ok, sounds like you have it covered. Thanks.

closed 2024-08-17 by ktbarrett