cocotb/cocotb #3980: Consider removing data and name attributes of `Event`

ktbarrett, 2024-06-25:
One can set a data attribute when `set()`ing and can give the Event a name. These features aren't used anywhere in the current tests and is something the user can do with a companion variable fairly easily. Nor is this something that `asyncio` supports.

alexforencich, 2024-12-27:
This is used extensively in cocotbext-axi.  Deleting it without deprecating it first is a bit rude.

ktbarrett, 2024-12-27:
1.9 will receive the deprecation. We plan on doing a "here's everything that's deprecated" release of 1.9 to act as a way to know what to change for 2.0 for existing tests.

I think there was a misunderstanding with 1.X that anything in particular was public API because there was no real effort put into separating the interfaces. 2.X+ will be a far more seemless experience for everyone.

ktbarrett, 2024-12-27:
On second thought, I might reintroduce this. It isn't standing in the way of 2.0 changes. But it will be deprecated.

PR #4079 (2024-07-30): Remove the `data` field from `Event`
Also removes the corresponding `data` argument to `Event.set`.

Closes #3980.
PR #4319 (2024-12-27): Reintroduce and deprecate `Event.data`
Closes #3980.

Reintroduce and deprecate the `data` field in `cocotb.triggers.Event` and the corresponding argument to `Event.set` that was removed in #4079. This is in the way of a 2.0 release, but it's definitely something we don't want to support going forward (`asyncio` API compatibility).
closed 2024-12-27 by ktbarrett