cocotb/cocotb #3989: pyuvm being tested?

raysalemi, 2024-06-27:
Hi @ktbarrett 

Does **cocotb** run tests with **pyuvm** it its regression suite?

Ray

ktbarrett, 2024-06-27:
We have a cocotb ecosystem compatibility test suite, but currently it only includes cocotb-coverage. https://github.com/cocotb/cocotb/actions/runs/9688874831/workflow

I think adding pyuvm would be reasonable. But it's currently not in a state to be tested.

raysalemi, 2024-06-27:
When you say "it" do you mean pyuvm or the version of cocotb?

I'd be interested in getting it into a testable state and having it incorporated.  I am on summer vacation.

ktbarrett, 2024-06-27:
"It" being the two working together. 

The ecosystem tests are there to ensure we aren't breaking things before a minor release is done. Since we are doing a major release, breaks are expected these tests are actually not valuable. But thinking about it again, I guess that means it's probably fine to add the test now.

raysalemi, 2024-06-27:
Please let me know what to do.

raysalemi, 2024-07-11:
Hi Kaleb,

I see you made pyuvm compatible with cocotb 2.0.  What are the next steps to include pyuvm in cocotb testing?  Please let me know what to do.

Ray

ktbarrett, 2024-07-11:
A test will need to be added to `.github/workflows/ecosystem-compat.yml`.

raysalemi, 2024-07-11:
Should I monkey-see-monkey-do that and submit a PR?

ktbarrett, 2024-07-11:
Yeah I think you could probably get away with just copy-paste-modifying the cocotb-coverage test in there.

closed 2024-07-23 by ktbarrett