cocotb/cocotb #3999: TB Never rives Signals

EdwardStables, 2024-07-01:


ktbarrett, 2024-07-01:
cocotb 1.8 only supports Verilator v5.006 or newer. https://docs.cocotb.org/en/stable/simulator_support.html#verilator

Use Verilator v5.024, as the newest release has broken cocotb support until we release cocotb 1.9 sometimes in the next week.

EdwardStables, 2024-07-01:
I saw the issue with the latest verilator version in other issues. I've tried verilator v5.024 with cocotb 1.8 and get the same result. But good to know v1.9 will be out in the near future.

```
(venv) ed@edpc:~/repos/vpu$ make
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/home/ed/repos/vpu'
rm -f results.xml
MODULE=sim.top_tb TESTCASE= TOPLEVEL=comb TOPLEVEL_LANG=verilog \
         build/Vtop 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:105  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/ed/repos/vpu/venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.024 2024-04-05
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /home/ed/repos/vpu/venv/lib/python3.10/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1719869319
     0.00ns INFO     cocotb.regression                  Found test sim.top_tb.test_sanity
     0.00ns INFO     cocotb.regression                  running test_sanity (1/1)
     0.00ns INFO     cocotb.comb                        a is 0000
     0.00ns INFO     cocotb.regression                  test_sanity failed
                                                        Traceback (most recent call last):
                                                          File "/home/ed/repos/vpu/sim/top_tb.py", line 8, in test_sanity
                                                            assert dut.c.value == 0b1111
                                                        AssertionError: assert 0000 == 15
                                                         +  where 0000 = ModifiableObject(comb.c).value
                                                         +    where ModifiableObject(comb.c) = HierarchyObject(comb).c
     0.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** sim.top_tb.test_sanity         FAIL           0.00           0.00          1.57  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                  0.00           0.03          0.03  **
                                                        **************************************************************************************
                                                        
- :0: Verilog $finish
make[1]: Leaving directory '/home/ed/repos/vpu'
(venv) ed@edpc:~/repos/vpu$ pip3 freeze | grep coco
cocotb==1.8.1
(venv) ed@edpc:~/repos/vpu$ verilator --version
Verilator 5.024 2024-04-05 rev v5.024
```

mmichilot, 2024-07-02:
With cocotb, values don't propagate through the design automatically because simulation is paused when executing Python code. Once the input values are set, simulation can be resumed by using a trigger, such as `Timer`

```systemverilog
@cocotb.test()
async def test_sanity(dut):
    dut.a.value = 0b1010
    dut.b.value = 0b0101
    await Timer(5) // Analogous to a #5 in Verilog
    dut._log.info("a is %s", dut.a.value)
    assert dut.c.value == 0b1111
```

EdwardStables, 2024-07-08:
Thanks for the suggestion, seems I oversimplified my example. But I'm getting this same behavior when using timers/clocks/etc. I've had cocotb testbenches working correctly previously and was trying to get this set up on a new machine, but have run into this issue.

EdwardStables, 2024-07-08:
I've just reverted back all my test changes to the full TB and things are now working as expected. I have no idea what the actual issue was here, I haven't actually changed anything in the environment.

Thank you both for your time, it's much appreciated!

closed 2024-07-08 by EdwardStables