cocotb/cocotb #4027: Common build arguments are not sufficient for multi-language projects

olagrottvik, 2024-07-11:
It would be useful for multilanguage projects to be able to separate `COMPILE_ARGS` into `COMPILE_VHDL_ARGS` and `COMPILE_VERILOG_ARGS.` E.g, the argument `-2008` is a valid VCOM argument for Questa, while for VLOG it would cause an error - meaning the user is forced to use legacy variables like `VCOM_ARGS` - and losing cross-simulator functionality.

Naturally, this is useful only for simulators which have support for both languages, like Questa, Riviera-PRO, etc.

ktbarrett, 2024-07-11:
I think the Makefiles are at the end of the road wrt further feature development. The Python runner already handles this by allowing the user to tag `build_args` with `VHDL` or `Verilog`.

olagrottvik, 2024-07-11:
Ok, this might be a feature of cocotb that I am unaware of. Apologies.

closed 2024-07-18 by ktbarrett