cocotb/cocotb #4035: Not able to access verilog from vhdl top using cocotb and Riviera PRO

leoeltipo, 2024-07-12:
Hello,

I have a very basic test bench with a vhdl top and two sub-modules: one vhdl and one verilog.

When trying to access the verilog submodule from a cocotb TB I have an error:

Unable to create gain via any registered implementation.

gain is the name of my sub-module. I traced this down to cocotb/lib/gpi/GpiCommon.cpp line 392. The one thing that confuses me is than even adding COCOTB_LOG_LEVEL=DEBUG the previous print in line 387 does not shows up...

Anyway this is the problem. I'm using:
* cocotb 1.8.1
* Python 3.10
* Riviera PRO simulator
* Linux
* I'm using a conda environment.

In the past, we solved the through-hierarchy issue (Modelsim) by adding _discover_all(), but that is the exact function throwing the error now and the issue is not solved. 

Any help would be very welcomed.

ktbarrett, 2024-07-12:
Can you provide an example?

leoeltipo, 2024-07-12:
Ok I'm working on packing the example. In the meantime, just was able to add some more debug and I get this:

# COUT:   1000.00ns DEBUG    gpi                                Searching for gain_i
# COUT:   1000.00ns DEBUG    gpi                                VHPI: vhpiVerilog returned from vhpi_get(vhpiType, ...)
# COUT:   1000.00ns DEBUG    gpi                                VHPI: Unable to fetch object top.gain_i
# COUT:   1000.00ns DEBUG    gpi                                Already checked VHPI implementation
# COUT:   1000.00ns DEBUG    gpi                                Checking if gain_i is native through implementation VPI
# COUT:   1000.00ns DEBUG    gpi                                Unable to query vpi_get_handle_by_name top.gain_i

That gain_i is the block I can't access from the cocotb test

ktbarrett, 2024-07-12:
I mean a *whole* example: HDL code, cocotb test, Makefile/Python runner, etc. Everything we need to reproduce the error.

leoeltipo, 2024-07-12:
[example.zip](https://github.com/user-attachments/files/16197640/example.zip)

leoeltipo, 2024-07-15:
Were you able to reproduce the error with that example?

leoeltipo, 2024-07-17:
Anybody can help with this issue?

leoeltipo, 2024-07-19:
By adding a bit more debug I found this:

VHPI: vhpi_scan found counter_i (1024) kind:vhpiCompInstStmtK name:counter_i
VHPI: vhpi_scan found gain_i (1117) kind:(null) name:gain_i

The first line is a vhdl submodule found ok. The second line is the verilog submodule. The (null) kind is returned by Riviera PRO vhpi_scan...not clear if this is expected

ktbarrett, 2024-07-19:
Based on what you posted before, that's expected.
```
# COUT:   1000.00ns DEBUG    gpi                                VHPI: vhpiVerilog returned from vhpi_get(vhpiType, ...)
# COUT:   1000.00ns DEBUG    gpi                                VHPI: Unable to fetch object top.gain_i
# COUT:   1000.00ns DEBUG    gpi                                Already checked VHPI implementation
# COUT:   1000.00ns DEBUG    gpi                                Checking if gain_i is native through implementation VPI
# COUT:   1000.00ns DEBUG    gpi                                Unable to query vpi_get_handle_by_name top.gain_i
```

The issue is it can't find the Verilog entity after switching to a VPI lookup. There's a million and one reasons why that could be.

leoeltipo, 2024-07-19:
I don't know if this adds something, but when simulating with Modelsim this is what I get for the DEBUG of this discover all section:

#   1000.00ns DEBUG    gpi                                Attempting to create GPI object from handle (Type=20, FullType=226).
#   1000.00ns DEBUG    gpi                                Handle is not a VHDL type.
#   1000.00ns DEBUG    gpi                                Found a name but unable to create via native implementation, trying others
#   1000.00ns DEBUG    gpi                                Searching for gain_i
#   1000.00ns DEBUG    gpi                                Skipping FLI implementation
#   1000.00ns DEBUG    gpi                                Checking if gain_i is native through implementation VPI
#   1000.00ns DEBUG    gpi                                VPI: Created GPI object from type vpiModule(32)
#   1000.00ns DEBUG    gpi                                Found gain_i via VPI
#   1000.00ns DEBUG    gpi                                Checking /top.gain_i exists
#   1000.00ns DEBUG    cocotb.gain_i                      Created

So the GPI prints it's not a VHDL type and is not a native implementation, and the it's the VPI that creates the object...something similar should happen with Riviera PRO I guess?

ktbarrett, 2024-07-19:
Yes, that's how mixed language discovery currently works in cocotb:
1. Use the interface of the parent object to attempt to discover the child object
2. If not fall back to global lookup
3. Use the first interface to attempt to look up the object (this is the interface of the toplevel, so in your case it's the VHPI again)
4. If that fails go to the next impl (VPI in your case)

The VPI look up *should* find the object, but isn't for whatever reason. cocotb is behaving as expected, but it isn't finding the object. That could be Riviera's fault or it could be cocotb's (at least when working with Riviera).

You could attempt to discover the object by looking up the object yourself:
```python
gain_i_handle = cocotb.SimHandle(cocotb.simulator.get_root_handle("/top.gain_i"))
```

That should allow you to go forward and prove that the issue is cocotb's mixed language discovery logic. If you can't find it, the issue may be Riviera not exposing that object for whatever reason.

leoeltipo, 2024-07-19:
Thanks for the clarification. I tried that but I get this error:

File "/home/stefanl/workspace/mcrfpga/main_fpga/test/test_tmp.py", line 66, in tb
         gain_i_handle = cocotb.SimHandle(cocotb.simulator.get_root_handle("/top.gain_i"))
         AttributeError: module 'cocotb' has no attribute 'SimHandle'

Am I missing an import?

leoeltipo, 2024-07-19:
Ok I just removed the cocotb.SimHandle and left SimHandle...anyway I get this error:

WARNING  root:cocotb_support.py:82   1000.00ns WARNING  gpi                                VHPI: DUT ':' doesn't match requested toplevel /top.gain_i
INFO     root:cocotb_support.py:86   1000.00ns INFO     gpi                                Nothing visible via VPI
ERROR    root:cocotb_support.py:80   1000.00ns ERROR    gpi                                No root handle found
ERROR    root:cocotb_support.py:80   1000.00ns INFO     cocotb.regression                  tb failed
                                                        Traceback (most recent call last):
                                                          File "/home/stefanl/workspace/mcrfpga/main_fpga/test/test_tmp.py", line 66, in tb
                                                            gain_i_handle = SimHandle(cocotb.simulator.get_root_handle("/top.gain_i"))
                                                          File "/home/stefanl/workspace/mcrfpga/main_fpga/cenv_pysim/lib/python3.8/site-packages/cocotb/handle.py", line 1155, in SimHandle
                                                            t = handle.get_type()
                                                        AttributeError: 'NoneType' object has no attribute 'get_type'
INFO     root:cocotb_support.py:86   1000.00ns INFO     cocotb.regression                  ****************************

Just want to check with you if this is effectively the VPI not doing what it should. Thanks a lot again.

ktbarrett, 2024-07-19:
Yeah it looks like Riviera's VPI isn't finding the object. More debugging will be needed.

leoeltipo, 2024-07-19:
Just to double check it the object hierarchy specification right? If I try with the object that is found (VHDL) as example it also fails. 

Im doing:

gain_i_handle = SimHandle(cocotb.simulator.get_root_handle("/top.counter_i"))

but top.counter_is actually found properly. I'm wondering if it's just a problem of how the hierarchy separator is used in riviera...

marlonjames, 2024-07-19:
This may call for a custom VPI application to test for what is visible through the interface with your design.

leoeltipo, 2024-07-19:
ah ok ...if you have an example of a custom VPI application I can try not very familiar on how to do that

marlonjames, 2024-07-19:
You could start with https://github.com/marlonjames/vpi-experiments/ and make your own based on `int_param_check.c` that instead does some hierarchy lookups.

Here is where we look for the top-level modules:
https://github.com/cocotb/cocotb/blob/d545d6eca22d8ca15acde60f27a234891acad80d/src/cocotb/share/lib/vpi/VpiImpl.cpp#L566

Here is where we are looking up by name:
https://github.com/cocotb/cocotb/blob/d545d6eca22d8ca15acde60f27a234891acad80d/src/cocotb/share/lib/vpi/VpiImpl.cpp#L324