cocotb/cocotb #4045: error: cannot load VPI module

zeidph, 2024-07-16:
Hi,

Cocotb seems like a very powerful framework for vhdl simulation but unfortunately i cant get to install it properly. I keep getting the following error: **cannot load VPI module** .

I followed the 6 steps under "Windows installation using MSYS2" found here :https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions . I followed the 64-bit options in step 4 and 6

### the cocotb version used:
Cocotb version: 1.9.0
[image]

### the operating system and version (32/64 bit)
64-bit Windows

### the simulator and version (32/64 bit)
GHDL version 4.1.0 (should be 64-bit but i am not sure how to check that?)
[image]

### the Python version, and where it's coming from
3.11.9, installed using msys2
[image]

### (part of) the log file (cleaned of proprietary information)
[image]

counter.vhd
```
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.STD_LOGIC_UNSIGNED.ALL;

entity UP_COUNTER is
    Port ( clk: in std_logic; -- clock input
           reset: in std_logic; -- reset input 
           counter: out std_logic_vector(3 downto 0) -- output 4-bit counter
     );
end UP_COUNTER;

architecture Behavioral of UP_COUNTER is
signal counter_up: std_logic_vector(3 downto 0);
begin
-- up counter
process(clk)
begin
if(rising_edge(clk)) then
    if(reset='1') then
         counter_up <= x"0";
    else
        counter_up <= counter_up + x"1";
    end if;
 end if;
end process;
 counter <= counter_up;

end Behavioral;
```
test_my_design.py
```
import cocotb
from cocotb.triggers import FallingEdge, Timer

async def generate_clock(dut):
    """Generate clock pulses."""

    for cycle in range(10):
        dut.clk.value = 0
        await Timer(1, units="ns")
        dut.clk.value = 1
        await Timer(1, units="ns")

@cocotb.test()
async def my_second_test(dut):
    """Try accessing the design."""

    await cocotb.start(generate_clock(dut))  # run the clock "in the background"

    await Timer(5, units="ns")  # wait a bit
    await FallingEdge(dut.clk)  # wait for falling edge/"negedge"
```

test_runner.py
```
# This file is public domain, it can be freely copied without restrictions.
# SPDX-License-Identifier: CC0-1.0

# test_runner.py

import os
from pathlib import Path

from cocotb.runner import get_runner

def test_my_design_runner():
    sim = os.getenv("SIM", "ghdl")

    proj_path = Path(__file__).resolve().parent

    sources = [proj_path / "counter.vhd"]

    runner = get_runner(sim)
    runner.build(
        vhdl_sources=sources,
        hdl_toplevel="UP_COUNTER",
        build_args=["-fsynopsys"],
        verbose = True,
    )
    runner.test(hdl_toplevel="UP_COUNTER", test_module="test_my_design,", test_args=["-fsynopsys"], verbose = True)

if __name__ == "__main__":
    test_my_design_runner()
```

zeidph, 2024-07-16:
I tried to run the simulation using a makefile but that didn't work either
```
zp@ID2021583 MINGW64 /c/cocotb_test
$ ls
Makefile  counter.vhd  test_my_design.py  test_runner.py

zp@ID2021583 MINGW64 /c/cocotb_test
$ make
rm -f results.xml
"make" -f Makefile results.xml
make[1]: Entering directory '/c/cocotb_test'
mkdir -p sim_build
\
/mingw64/bin/ghdl -i -fsynopsys  --workdir=sim_build --work=work /c/cocotb_test/counter.vhd && \
/mingw64/bin/ghdl -m -fsynopsys  --workdir=sim_build -Psim_build --work=work UP_COUNTER
analyze C:/cocotb_test/counter.vhd
elaborate up_counter
rm -f results.xml
MODULE=test_my_design TESTCASE= TOPLEVEL=UP_COUNTER TOPLEVEL_LANG=vhdl \
 /mingw64/bin/ghdl -r -fsynopsys  --workdir=sim_build -Psim_build --work=work UP_COUNTER  --vpi=C:/msys64/mingw64/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.dll
loading VPI module 'C:/msys64/mingw64/lib/python3.11/site-packages/cocotb/libs/libcocotbvpi_ghdl.dll'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.11/site-packages/pygpi/entry.py", line 19, in load_entry
    entry_module = importlib.import_module(entry_module_str)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 944, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:/msys64/mingw64/lib/python3.11/site-packages/cocotb/__init__.py", line 43, in <module>
    import cocotb.handle
  File "C:/msys64/mingw64/lib/python3.11/site-packages/cocotb/handle.py", line 32, in <module>
    import ctypes
  File "C:/msys64/mingw64/lib/python3.11/ctypes/__init__.py", line 8, in <module>
    from _ctypes import Union, Structure, Array
ImportError: DLL load failed while importing _ctypes: The specified module could not be found.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.11/site-packages/pygpi/entry.py", line 22, in load_entry
    raise RuntimeError(
RuntimeError: Failure to parse PYGPI_ENTRY_POINT ('cocotb:_initialise_testbench')
ERROR: results.xml was not written by the simulation!
make[1]: *** [C:/msys64/mingw64/lib/python3.11/site-packages/cocotb/share/makefiles/simulators/Makefile.ghdl:106: results.xml] Error 1
make[1]: Leaving directory '/c/cocotb_test'
make: *** [C:/msys64/mingw64/lib/python3.11/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```
Makefile
```
# Makefile

# defaults
SIM ?= ghdl
TOPLEVEL_LANG ?= vhdl

VHDL_SOURCES += $(PWD)/counter.vhd
# use VHDL_SOURCES for VHDL files

# TOPLEVEL is the name of the toplevel module in your Verilog or VHDL file
TOPLEVEL = UP_COUNTER

# MODULE is the basename of the Python test file
MODULE = test_my_design

EXTRA_ARGS = -fsynopsys

# include cocotb's make rules to take care of the simulator setup
include $(shell cocotb-config --makefiles)/Makefile.sim
```

ktbarrett, 2024-07-16:
Windows support has been problematic for forever. *All* POSIX-like environments for Windows that are needed by cocotb at the moment have issues with cocotb (and issues in general). Every time we fix them new issues pop up. Going forward on Windows we are going to try to get native Windows operation to work (no POSIX compat layer), but that work hasn't been done yet.

I would suggest for now instead of attempting to use MSYS2 or any similar system to instead use Linux directly or using it via Windows Subsystem for Linux, and follow the normal installation instructions for Linux.

ktbarrett, 2024-07-16:
Closing as this is a known issue and the given approach is deprecated. (I'll update the Wiki page).

closed 2024-07-16 by ktbarrett

marlonjames, 2024-07-17:
You may have better luck installing and running cocotb from a normal Windows Python environment.

```
python -m venv env
env\Scripts\activate
python -m pip install cocotb
python test_runner.py
```

Gophen, 2024-10-28:
> You may have better luck installing and running cocotb from a normal Windows Python environment.
> 
> ```
> python -m venv env
> env\Scripts\activate
> python -m pip install cocotb
> python test_runner.py
> ```

Unfortunately, running cocotb from a normal Python Windows venv didn't solve the issue.
GHDL has to be installed inside a POSIX environment, and adding support for that in cocotb would be a nightmare, as stated by ktbarrett.