cocotb/cocotb #4047: Version 1.9.0 breaks my simulation

jefflieu, 2024-07-17:
Latest update of cocotb breaks our simulation and I can't figure it out yet. It's related to Xilinx's `glbl.v` module.
This is our Makefile:
```
EXTRA_TOPS ?= work.glbl

TOPLEVEL_LANG ?= vhdl

TOPLEVEL ?= ...
MODULE   ?= ...
include $(shell cocotb-config --makefiles)/Makefile.sim

#Note that this must compile into the same library as everything else
#TODO: Make this better - find an appropriate variable, or find a way to include it in the flow.
work.glbl:
        vlib sim_build/work
        vlog -work sim_build/work ${XIL_DIR}/Vivado/${VIVADO_REVISION}/data/verilog/src/glbl.v
````
The error is:

```
# VSIM: Error: Unknown library unit 'glbl' specified.
# VSIM: Error: Simulation initialization failed.
```

What has changed that breaks this area of compilation?

jefflieu, 2024-07-17:
It seems to be this change in runsim.tcl
````
onerror {
        puts [read [open sim.log r]]
        quit -code 1
}
amap -c
@if [string length [array get env LICENSE_QUEUE]] {
 set LICENSE_QUEUE $::env(LICENSE_QUEUE)
}
if [file exists sim_build/work] {adel -lib sim_build/work -all}
alib sim_build/work
amap work sim_build/work
set worklib work
```
which would clear the library I think.

ktbarrett, 2024-07-17:
Have you considered integrating against the change by using `VHDL_SIM_ORDER` and `VHDL_SOURCES_$(LIB)`?

jefflieu, 2024-07-17:
The file is verilog file. I managed work around by compiling that into a separate library. Like this:
```
glbl:
        vlib sim_build/verilog_lib
        vlog -work sim_build/verilog_lib ${XIL_DIR}/Vivado/${VIVADO_REVISION}/data/verilog/src/glbl.v
```
Using `VERILOG_SOURCES=glbl.v` doesn't work because we have `-2008` as COMPILE_ARGS and it's not valid for verilog files.
This is no longer related to the original issue but how is it supposed to be done when you have mixture of VHDL and Verilog code, using flow with Makefile?

ktbarrett, 2024-07-17:
Ah, I see. Yeah libraries aren't limited to VHDL in mixed language sims. Must `glbl` be in the library called `work` to function correctly? You *might* be able to compile it into a different library. Alternatively, you could change the toplevel library name for the rest of the sources (by setting `RTL_LIBRARY`), it doesn't have to be called `work` (nor should it really...).

jefflieu, 2024-07-18:
The gist of the issue is that:
* We need to compile some verilog files separately, because "VERILOG_SOURCES" doesn't work, because our COMPILE_ARGS has `-2008` in it. 
* So what we did was to compile this Verilog glbl.v first into work library as a separate step, done before the make sim which then runs `compile all other VHDL files` 
* This worked so far until recent change wipes out the `work` library just before the VHDL compilation --> which is why I lost the glbl module.
---> So the work around is to compile glbl.v file into a different library.  Luckily this file is just for simulation only, we don't directly instantiate them.

Thanks for pointing to the library, this is resolved for me.

closed 2024-07-18 by jefflieu