cocotb/cocotb #4059: #1076 breaks using xcelium older than 23.09

samimia-swks, 2024-07-20:
With cocotb 1.9.0, due to #1076, if using xcelium that's older than 23.09, xcelium errors out with 
Unkown option NEW_VHPI_PROPAGATE_DELAY

The option should only be used for xcelium versions that support it...

ktbarrett, 2024-07-24:
Pinging @imphil. How much of a pain would it be to use a version ifdef?

imphil, 2024-07-28:
@samimia-swks The define `NEW_VHPI_PROPAGATE_DELAY` is only set for VHDL designs. Before Xcelium introduced this flag, VHDL designs did not work at all, or at least failed at very basic tasks. Hence the minimum supported version for the combination cocotb/Xcelium/VHPI is a 23.09 (or an older version that has this flag as well; have a look at the release notes for the version you're using if the fix is available there).

Were you running cocotb 1.8 against an Xcelium/VHDL design that did not show buggy behavior before?

samimia-swks, 2024-07-28:
> CAUTION: The email below is from an external source. Use caution before opening attachments or clicking links.

My design is verilog only.
So it seems the bug is that the switch should not be passed if the design is verilog only…

Arman Samimi

imphil, 2024-07-28:
So I'm assuming you're using the cocotb runner? I cannot see a problem in the Makefiles.

Please provide a bit more information about what you're doing exactly to reproduce this issue.

samimia-swks, 2024-08-01:
@imphil yes I am using the python runner, invoked as below. I specify `hdl_toplevel_lang="verilog"` and pass only .v and sv files to `verilog_sources`
``` 
      runner = cocotb.runner.get_runner("xcelium")
      runner.build(
          hdl_library="top",
          verilog_sources=verilog_sources,
          includes=includes,
          defines=defines,
          parameters={},
          build_args=build_args,
          hdl_toplevel=top_level,
          always=False,
          build_dir=sim_root,
          verbose=False,
      )
      runner.test(
          hdl_toplevel=top_level,
          testcase=coco_seq,
          test_module=test_name,
          hdl_toplevel_library="top",
          hdl_toplevel_lang="verilog",
          gpi_interfaces=None,
          seed=None,
          test_args=test_args,
          plusargs=[],
          extra_env=extra_env,
          waves=False,
          gui=False,
          parameters=None,
          build_dir=sim_root,
          test_dir=sim_root,
          results_xml="results.xml",
          verbose=False,
      )
````

commit eb045fb7 (2024-08-01): Fix #4059

Guards addition of `-NEW_VHPI_PROPAGATE_DELAY` on whether VHDL sources are present.
PR #4085 (2024-08-05): Guard VHDL-specific option in Xcelium Python runner.
Closes #4059. Guards addition of `-NEW_VHPI_PROPAGATE_DELAY` on whether VHDL sources are present.
closed 2024-08-05 by ktbarrett