cocotb/cocotb #4065: virtual env not found when installed with pipx

mbuechl, 2024-07-23:
The scenario is this: I have a rather simple `Makefile_test_something` containing `SIM=modelsim`, and I am using "Questa Sim-64 Version 2022.4_2 linux_x86_64 Dec  9 2022".

Since upgrading to Ubuntu 24.04, installing Python packages in a user-wide way with `pip install cocotb` did not seem to work anymore, so I used [pipx](https://github.com/pypa/pipx) instead.

Now `cocotb-config` is a symlink in `~/.local/bin/cocotb-config` that points at `.local/share/pipx/venvs/cocotb/bin/cocotb-config` containing this:
```
#!/home/mbuechl/.local/share/pipx/venvs/cocotb/bin/python
# -*- coding: utf-8 -*-
import re
import sys
from cocotb.config import main
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(main())
```

So there still is a venv, it is just conveniently hidden from me.

If I had used a manually managed venv I would have to `source some_path/venv/bin/activate` before running the simulation, which sets the `VIRTUAL_ENV` environment variable. This would then be picked up by the code that I linked below, and all would work. https://github.com/cocotb/cocotb/blob/221fefd3e7997ed09c5e7bff228496f0db1c4f7d/src/cocotb/share/lib/embed/gpi_embed.cpp#L77

But `VIRTUAL_ENV` is not set, so I get the corresponding error message in my Questa log:

```
#      -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
```

and in the terminal: `ModuleNotFoundError: No module named 'pygpi'`

Workarounds:
- manually set up and source a virtual env for cocotb
- manually do `export VIRTUAL_ENV=/home/mbuechl/.local/share/pipx/venvs/cocotb`

I made a quick and dirty patch to set `VIRTUAL_ENV` in the Makefiles, which works.

I could also imagine that just adapting the documentation at https://docs.cocotb.org/en/stable/install.html#installation-of-cocotb would be an option.

```diff
diff --git a/src/cocotb_tools/makefiles/Makefile.sim b/src/cocotb_tools/makefiles/Makefile.sim
index 4171bd54..268f7d42 100644
--- a/src/cocotb_tools/makefiles/Makefile.sim
+++ b/src/cocotb_tools/makefiles/Makefile.sim
@@ -101,6 +101,8 @@ SIM_LOWERCASE := $(shell echo $(SIM) | tr A-Z a-z)
 # Directory containing the cocotb Makfiles
 COCOTB_MAKEFILES_DIR := $(shell cocotb-config --makefiles)
 
+VIRTUAL_ENV := $(subst bin/python,,$(shell cocotb-config --python-bin))
+
 include $(COCOTB_MAKEFILES_DIR)/Makefile.deprecations
 
 HAVE_SIMULATOR = $(shell if [ -f $(COCOTB_MAKEFILES_DIR)/simulators/Makefile.$(SIM_LOWERCASE) ]; then echo 1; else echo 0; fi;)
diff --git a/src/cocotb_tools/makefiles/simulators/Makefile.questa b/src/cocotb_tools/makefiles/simulators/Makefile.questa
index 1ee18d87..0c5016ac 100644
--- a/src/cocotb_tools/makefiles/simulators/Makefile.questa
+++ b/src/cocotb_tools/makefiles/simulators/Makefile.questa
@@ -161,6 +161,7 @@ $(COCOTB_RESULTS_FILE): $(SIM_BUILD)/runsim.do
        set -o pipefail; COCOTB_TEST_MODULES=$(call deprecate,MODULE,COCOTB_TEST_MODULES) COCOTB_TESTCASE=$(call deprecate,TESTCASE,COCOTB_TESTCASE) COCOTB_TEST_FILTER=$(COCOTB_TEST_FILTER) COCOTB_TOPLEVEL=$(call deprecate,TOPLEVEL,COCOTB_TOPLEVEL) \
        GPI_EXTRA=$(GPI_EXTRA) TOPLEVEL_LANG=$(TOPLEVEL_LANG) \
        $(SIM_CMD_PREFIX) $(CMD) $(RUN_ARGS) -do $(SIM_BUILD)/runsim.do $(call deprecate,PLUSARGS,COCOTB_PLUSARGS) $(SIM_CMD_SUFFIX)
+       VIRTUAL_ENV=$(VIRTUAL_ENV) \
 
        $(call check_for_results_file)
```

ktbarrett, 2024-07-23:
Also you can work around the problem using `pip install -U cocotb`. Or using any virtualenv manager out there: `virtualenv`, `venv`, `pyenv`, `conda`, etc. etc.

closed 2024-07-24 by ktbarrett

mbuechl, 2024-07-24:
> Dupe of #3643.

Oh! Looks obvious to find but somehow I missed this.

> If you want to put up a PR that passes `cocotb-config --python-bin` in on `COCOTB_PYTHON_BIN` envvar and remove the `PYTHON_INTERPRETER_PATH` stuff, I'd appreciate it.

You already opened a merge request while I was asleep, which is great. I tested it and for me it works nicely, either in a manually created `venv` and with `pipx install ~/src/cocotb/cocotb/`.

Thanks a lot for looking into this.

mbuechl, 2024-07-24:
> Also you can work around the problem using `pip install -U cocotb`. Or using any virtualenv manager out there: `virtualenv`, `venv`, `pyenv`, `conda`, etc. etc.

More for the sake of completeness, here is how Ubuntu 24.04 complains when I try `pip install cocotb`:

```
$ pip install -U cocotb
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

ktbarrett, 2024-07-24:
Well that's slightly annoying. Ubuntu used to set up a user Python environment in `~/.local/`.