cocotb/cocotb #4074: FST trace not working with verilator

jhladik, 2024-07-26:
I have run into issues with generating .fst trace with cocotb and ventilator. Note that the .vcd trace generates fine.

Environment:
* cocotb version 1.9.0
* macOS 14.5 Darwin Kernel Version 23.5.0: Wed May  1 20:19:05 PDT 2024; root:xnu-10063.121.3~5/RELEASE_ARM64_T8112 arm64
* Verilator 5.027 devel rev v5.026-82-gfab268f77
* brew-installed python3 Python 3.12.4)

`
verilator -cc --build --exe -j 0 -Mdir build -Wall -DCOCOTB_SIM=1 -DVM_TRACE_FST \
                --vpi --public-flat-rw --prefix Vtop -o Vtop \
                -LDFLAGS \
                        "-Wl,-rpath,`cocotb-config --lib-dir` -L`cocotb-config --lib-dir` \
                        -lcocotbvpi_verilator" \
                --trace --trace-fst --trace-structs --timescale 1ns/1ps \
                `cocotb-config --share`/lib/verilator/verilator.cpp uart_protocol.sv

PYTHONDONTWRITEBYTECODE=1 \
                MODULE=uart_protocol_tb \
                TESTCASE= \
                TOPLEVEL=uart_protocol \
                TOPLEVEL_LANG=verilog \
                build/Vtop --trace --trace-fst --trace-structs -DVM_TRACE_FST
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:109  in set_program_name_in_venv        Using Python virtual environment interpreter at /Users/jhladik/Downloads/Testing/.venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.027 devel
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.0 from /Users/jhladik/Downloads/Testing/.venv/lib/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1722013637
     0.00ns INFO     cocotb.regression                  Found test uart_protocol_tb.test_initial_outputs
     0.00ns INFO     cocotb.regression                  Found test uart_protocol_tb.test_write_no_backpressure
     0.00ns INFO     cocotb.regression                  Found test uart_protocol_tb.test_read_no_backpressure
     0.00ns INFO     cocotb.regression                  Found test uart_protocol_tb.test_write_backpressure
     0.00ns INFO     cocotb.regression                  Found test uart_protocol_tb.test_read_backpressure
     0.00ns INFO     cocotb.regression                  Found test uart_protocol_tb.test_reset_outputs
     0.00ns INFO     cocotb.regression                  running test_initial_outputs (1/6)
Assertion failed: (hierarchicalName.rfind(' ') != std::string::npos), function declare, file verilated_fst_c.cpp, line 169.
make: *** [dump.fst] Abort trap: 6
make: *** Deleting file 'dump.fst'`

I can generate an .fst trace with Verilator using a SystemVerilog test bench without cocotb ran with the following commands:
`verilator --binary --timing --exe --trace --trace-fst --trace-structs --x-assign unique -j 0 -Wall uart_protocol_tb.sv uart_protocol.sv
obj_dir/Vuart_protocol_tb`.

I believe this issue is related to the following Verilator pull request: https://github.com/verilator/verilator/pull/4612

ktbarrett, 2024-07-26:
What is cocotb doing or not doing to break this that was changed upstream?

ktbarrett, 2024-07-26:
And if you think an upstream change broke this, why did you report the issue here?

jhladik, 2024-07-26:
Sorry, I am not very knowledgeable on the functioning of verilator and cocotb. I am just suspecting the referenced verilator change which happened in the past 8 months broke the .fst generation with cocotb.

I am looking for help investigating the root cause of the issue.

jhladik, 2024-07-26:
Here is an example with a Makefile which demonstrates the issue: [simplest.tar.gz](https://github.com/user-attachments/files/16395467/simplest.tar.gz)

`make vcd` generates a dump.vcd which has no issues.
`make fst` attempts to generate a dump.fst file but fails while producing it:

```bash
MODULE=simplest_tb TESTCASE= TOPLEVEL=simplest TOPLEVEL_LANG=verilog sim_build/Vtop --trace
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:109  in set_program_name_in_venv        Using Python virtual environment interpreter at /Users/jhladik/Downloads/Testing/.venv/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.027 devel
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.0 from /Users/jhladik/Downloads/Testing/.venv/lib/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1722018467
     0.00ns INFO     cocotb.regression                  Found test simplest_tb.simplest_test
     0.00ns INFO     cocotb.regression                  running simplest_test (1/1)
Assertion failed: (hierarchicalName.rfind(' ') != std::string::npos), function declare, file verilated_fst_c.cpp, line 169.
make: *** [fst] Abort trap: 6
```

ktbarrett, 2024-07-29:
Seems that the `hierarchicalName` in the design I tested that it's complaining about is `clk`.

AndrewNolte, 2024-09-03:
I'd suggest building Verilator without that assertion. There's a comment that says the lastWord function assumes that the string has a space, but it seems to handle it. If it works, then you can put up a PR there.

AndrewNolte, 2024-09-03:
Actually it's a Verilator include, so it shouldn't require rebuilding verilator, just the verilator.cpp in cocotb with the updated `include/verilated_fst_c.cpp`

imphil, 2024-09-04:
@AndrewNolte Kaleb opened a Verilator PR, it's here: https://github.com/verilator/verilator/pull/5313

ktbarrett, 2024-09-24:
This was fixed upstream. Can you try it @jhladik?

jhladik, 2024-09-24:
> This was fixed upstream. Can you try it @jhladik?

It is working now. Tested with https://github.com/verilator/verilator/commit/48faf8d036db16573855d3eeb296b57de0ffca2a which is referenced in https://github.com/verilator/verilator/issues/5470. Thanks everyone for your support!

closed 2024-09-25 by imphil