cocotb/cocotb #4075: Unable to obtain correct output results despite module tested to work using only Verilator without cocotb

booth-algo, 2024-07-26:
cocotb version: 1.9.0 (also tested on 2.0.0.dev0+f756f744)
OS: WSL2 Ubuntu 22.04 (on Windows 10 64 bit)
simulator: Verilator 5.027 devel rev v5.026-22-gde44ca8df
Python: 3.11.9, from Conda, but also tested on WSL self-installed path

Issue:
I am learning how to integrate cocotb into my current project. I am able to run it successfully for a simple test adder.sv design with a testbench written following the documentation.
However, when I attempt to integrate cocotb for my actual project .sv modules, it asserts all zeroes for the result signal. I have verified using Verilator and its generated waveforms that this should not be the case.

Terminal log:
```terminal
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Verilator version 5.027 devel
     0.00ns INFO     cocotb                             Running tests with cocotb v1.9.0 from /home/---/.local/lib/python3.11/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1722032374
     0.00ns INFO     cocotb.regression                  Found test fpu_tb.test_fp_mul
     0.00ns INFO     cocotb.regression                  running test_fp_mul (1/1)
                                                          Test for floating-point multiplication
    90.00ns INFO     cocotb.fp_mul                      Input a: 001111111000000000000000
    90.00ns INFO     cocotb.fp_mul                      Input b: 001111111000000000000000
    90.00ns INFO     cocotb.fp_mul                      Result: 000000000000000000000000
   100.00ns INFO     cocotb.fp_mul                      Result at 10 cycles: 000000000000000000000000
   110.00ns INFO     cocotb.fp_mul                      Result at 11 cycles: 000000000000000000000000
   120.00ns INFO     cocotb.fp_mul                      Result at 12 cycles: 000000000000000000000000
   130.00ns INFO     cocotb.fp_mul                      Result at 13 cycles: 000000000000000000000000
   140.00ns INFO     cocotb.fp_mul                      Result at 14 cycles: 000000000000000000000000
   150.00ns INFO     cocotb.fp_mul                      Result at 15 cycles: 000000000000000000000000
   160.00ns INFO     cocotb.fp_mul                      Result at 16 cycles: 000000000000000000000000
   170.00ns INFO     cocotb.fp_mul                      Result at 17 cycles: 000000000000000000000000
   180.00ns INFO     cocotb.fp_mul                      Result at 18 cycles: 000000000000000000000000
   190.00ns INFO     cocotb.fp_mul                      Result at 19 cycles: 000000000000000000000000
   190.00ns INFO     cocotb.regression                  test_fp_mul failed
                                                        Traceback (most recent call last):
                                                          File "/home/---/tb/fpu_tb.py", line 36, in test_fp_mul
                                                            assert dut.result.value == expected_result, f"Error (fp mul): {dut.result.value} != {expected_result}"
                                                        AssertionError: Error (fp mul): 000000000000000000000000 != 4161536
                                                        assert 000000000000000000000000 == 4161536
                                                         +  where 000000000000000000000000 = ModifiableObject(fp_mul.result).value
                                                         +    where ModifiableObject(fp_mul.result) = HierarchyObject(fp_mul).result
   190.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** fpu_tb.test_fp_mul             FAIL         190.00           0.00      47368.16  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                190.00           0.04       4460.13  **
                                                        **************************************************************************************
                                                        
- :0: Verilog $finish
```

Certain paths have been replaced with "---".

Information:
This is testing a floating-point multiplication module. We run a floating-point 24 system to optimise DSP usage and fit the OpenGL ES2 standard. May I repeat that this has been tested to work using purely Verilator and checking generated waveforms, but isn't working with cocotb. 

Testbench:
```python
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge, Timer

@cocotb.test()
async def test_fp_mul(dut):
    """Test for floating-point multiplication"""

    # Start the clock
    clock = Clock(dut.clk, 10, units="ns")
    cocotb.start_soon(clock.start())

    # Initialize inputs
    dut.a.value = 0x3f8000  # 1.0 in IEEE 754 format
    dut.b.value = 0x3f8000  # 1.0 in IEEE 754 format
    # dut.opcode.value = 0b0000

    for _ in range(10):
        await RisingEdge(dut.clk)

    # Debugging: print the values of inputs and result
    dut._log.info(f"Input a: {dut.a.value}")
    dut._log.info(f"Input b: {dut.b.value}")
    dut._log.info(f"Result: {dut.result.value}")

    # Wait more if necessary
    for cycle in range(10):
        await RisingEdge(dut.clk)
        # dut._log.info(f"A at {cycle + 10} cycles: {dut.a.value}")
        # dut._log.info(f"B at {cycle + 10} cycles: {dut.b.value}")
        dut._log.info(f"Result at {cycle + 10} cycles: {dut.result.value}")

    expected_result = 0x3f8000  # Expected result in IEEE 754 format
    assert dut.result.value == expected_result, f"Error (fp mul): {dut.result.value} != {expected_result}"
```

ktbarrett, 2024-07-26:
Do you mind putting up your other project files? Makefile, HDL, etc?

booth-algo, 2024-07-26:
fp_mul top
```systemverilog
module fp_mul #(
    parameter WIDTH = 24
)(
    input logic clk, 
    input wire [WIDTH - 1:0] a,
    input wire [WIDTH - 1:0] b,

    output wire [23:0] result
);

    logic [16:0] product_o;
    logic zero_flag_o;
    logic sign_xor_o;
    logic [8:0] sum_exp_o;

    fp_mul_0 fp_mul_0_inst (
        .clk(clk),
        .a(a),
        .b(b),
        .product_o(product_o),
        .zero_flag_o(zero_flag_o),
        .sign_xor_o(sign_xor_o),
        .sum_exp_o(sum_exp_o)
    );

    fp_mul_1 fp_mul_1_inst (
        .clk(clk),
        .product_i(product_o),
        .zero_flag_i(zero_flag_o),
        .sign_xor_i(sign_xor_o),
        .sum_exp_i(sum_exp_o),
        .result_o(result)
    );
    
endmodule
```
fp_mul_0 (pipeline stage 1)
```systemverilog
module fp_mul_0 #(
    parameter WIDTH = 24
)(
    input logic clk,

    input  wire [WIDTH - 1:0] a,
    input  wire [WIDTH - 1:0] b,

    output logic [16:0] product_o,
    output logic zero_flag_o,
    output logic sign_xor_o,
    output logic [8:0] sum_exp_o
);
    // Extract fields from input
    logic sign1 = a[23];
    logic sign2 = b[23];

    logic [7:0] exp1 = a[22:15];
    logic [7:0] exp2 = b[22:15];

    logic mantissa_top_bit_1 = exp1==0 ? 1'b0 : 1'b1;
    logic mantissa_top_bit_2 = exp2==0 ? 1'b0 : 1'b1;

    logic [15:0] mant1 = {mantissa_top_bit_1, a[14:0]}; // implicit leading 1
    logic [15:0] mant2 = {mantissa_top_bit_2, b[14:0]}; 

    logic [31:0] product = mant1 * mant2; // 16-bit mantissa * 16-bit mantissa = 32-bit product

    always_ff @(posedge clk) begin
        // If either input number has a high-order bit of zero, then that input is zero and the product is zero.
        if (mant1[15] == 0 || mant2[15] == 0) begin
            zero_flag_o <= 1;
        end
        /*
            The output exponent is exp1+exp2-126 or exp1+exp2-127. 
            If the sums of the input exponents is less than 128, 
            then the exponent will underflow and the product is zero.
        */
        else if (exp1 + exp2 < 128) begin
            zero_flag_o <= 1;
        end else begin
            zero_flag_o <= 0;
        end

        product_o <= product[31:15]; // top 17 bits of product

        sign_xor_o <= sign1 ^ sign2;
        sum_exp_o <= exp1 + exp2;

    end
    
endmodule
```
fp_mul_1 (pipeline stage 2)
```systemverilog
module fp_mul_1 #(
    parameter WIDTH = 24
)(
    input  logic            clk,
    input  wire [16:0]      product_i,
    input  wire             zero_flag_i,
    input  logic            sign_xor_i,
    input  wire [8:0]       sum_exp_i,
    
    output logic [WIDTH - 1:0] result_o
);

    logic sign_o;
    logic [8:0] exp_o;
    logic [14:0] mant_o;

    always_ff @(posedge clk) begin
        // If both inputs are nonzero and the exponents don't underflow        
        if (!zero_flag_i) begin
            // Then if (mantissa1)x(mantissa2) has the high order-bit set, the top 9-bits of the product are the output mantissa and the output exponent is exp1+exp2-128.
            if (product_i[16] == 1) begin
                exp_o <= sum_exp_i - 126;
                mant_o <= product_i[15:1]; // top 23 bits of product
            end

            // Otherwise the second bit of the product will be set, and the output mantissa is the top 9-bits of (product)<<1 and the output exponent is exp1+exp2-129.
            else begin
                exp_o <= sum_exp_i - 127;
                mant_o <= product_i[14:0]; // top 23 bits of product shifted left by 1
            end

            // The sign of the product is (sign1)xor(sign2)
            sign_o <= sign_xor_i;

        end else begin
            // If either input number has a high-order bit of zero, then that input is zero and the product is zero.
            sign_o <= 0;
            exp_o <= 0;
            mant_o <= 0;
        end

    end

    assign result_o = exp_o > 254 ? 24'h7f8000 : {sign_o, exp_o[7:0], mant_o};
    
endmodule
```
runner.py file (wraps the Makefile)
```python
from os import getenv
from pathlib import Path
from cocotb.runner import get_runner
from typing import Any

def simulate_pass(
    project_dir: Path,
    module_params: dict[str, Any] = {},
    extra_build_args: list[str] = [],
    trace: bool = False,
):
    rtl_dir = project_dir / "rtl"
    sim_dir = project_dir / "sim" / "sim_build"

    fp_dir = rtl_dir / "core" / "fp"
    verilog_sources = list(fp_dir.glob("*.sv"))

    verilog_sources = [rtl_dir / "adder.sv"] + verilog_sources

    SIM = getenv("SIM", "verilator")

    top_module = "fp_mul"
    test_module = "fpu_tb"

    runner = get_runner(SIM)
    runner.build(
        verilog_sources=verilog_sources,
        includes=[rtl_dir],
        hdl_toplevel=top_module,
        build_args=[
            # Verilator linter is overly strict.
            # Too many errors
            # These errors are in later versions of verilator
            "-Wno-GENUNNAMED",
            "-Wno-WIDTHEXPAND",
            "-Wno-WIDTHTRUNC",
            # Simulation Optimisation
            "-Wno-UNOPTFLAT",
            # Signal trace in dump.fst
            *(["--trace-fst", "--trace-structs", "--trace"] if trace else []),
            "-prof-c",
            "--stats",
            "--assert",
            "-O2",
            "-build-jobs",
            "8",
            "-Wno-fatal",
            "-Wno-lint",
            "-Wno-style",
        ],
        parameters=module_params,
        build_dir=sim_dir,
    )
    runner.test(
        hdl_toplevel=top_module,
        test_module=test_module,
        results_xml="results.xml",
    )

# Sample usage:
if __name__ == "__main__":
    project_dir = Path(__file__).parent.parent
    module_params = {} # Add module parameters here
    simulate_pass(
        project_dir=project_dir,
        module_params=module_params,
        extra_build_args=[],
        trace=True # Enable trace to generate waveforms
    )
```

ktbarrett, 2024-07-26:
Have you tried with Icarus? It seem the code should be simple enough for it to compile.

booth-algo, 2024-07-26:
> Have you tried with Icarus? It seem the code should be simple enough for it to compile.

No I haven't because I'd like to run Verilator in the long term anyways considering its performance compared to Icarus. Plus I managed to get Verilator working for a simple example so I'm just quite skeptical of what exactly is failing to work for my project modules.

teobiton, 2024-07-29:
The tests are failing because signals in `fp_mul_0` are declared and assigned at the same time. Maybe VPI does not like this construct ? Separating declaration and assignment will solve your issue.

ktbarrett, 2024-07-29:
Running this with Icarus shows the issue. Your design doesn't work. Xs are coming out, likely due to something not being hooked up correctly. Verilator doesn't support Xs, so you see 0s.

Next time you have an issue, be ready to break out a fully featured simulator to debug

closed 2024-07-29 by ktbarrett

booth-algo, 2024-07-29:
> Running this with Icarus shows the issue. Your design doesn't work. Xs are coming out, likely due to something not being hooked up correctly. Verilator doesn't support Xs, so you see 0s.
> 
> Next time you have an issue, be ready to break out a fully featured simulator to debug

I am a bit confused by this - the design works fine with Verilator (without using cocotb). 

> The tests are failing because signals in `fp_mul_0` are declared and assigned at the same time. Maybe VPI does not like this construct ? Separating declaration and assignment will solve your issue.

Thank you for this comment - I separated all the declarations and assignments, and cocotb passed. It is weird because shouldn't it automatically run a VPI construct that supports declarations and assignments? I understand that [Verilator support in cocotb is currently experimental](https://docs.cocotb.org/en/stable/simulator_support.html), but what modifications could I make to the build arguments to specify VPI?

ktbarrett, 2024-08-01:
We actually just ran into this at work the other day. An assignment statement with a declaration is an initial value assignment, not a constant assignment. Verilator worked with this at some point, but doesn't now / doesn't work with coverage enabled. Odd. But yes, invalid syntax / intent.

booth-algo, 2024-08-02:
> We actually just ran into this at work the other day. An assignment statement with a declaration is an initial value assignment, not a constant assignment. Verilator worked with this at some point, but doesn't now / doesn't work with coverage enabled. Odd. But yes, invalid syntax / intent.

Thanks for the follow up.