cocotb/cocotb #4082: Can't set string variables using `vpiInertialDelay` in Questa

ktbarrett, 2024-07-30:
We get the following when attempting to write to a string variable using `vpiInertialDelay`.

We have to use the following logic to instead set string variables using `vpiNoDelay`, which works for some reason.
https://github.com/cocotb/cocotb/blob/49a9f19e31df31963cc5ad6ccf5c365dd0d5bbd0/src/cocotb/share/lib/vpi/VpiSignal.cpp#L205-L214

imphil, 2025-01-29:
Filed Siemens upstream [case 01403637](https://support.sw.siemens.com/de-DE/support-case/details/5c45e780-de80-11ef-96fd-c522faa4b3ba).

imphil, 2025-02-11:
Feedback from Siemens:

I've tested the tool behavior with vpiInertialDelay, and the value of the string gets updated as expected.

```sv
string foo;  // String variable
import "DPI-C" function int myTask(input string userdata);

initial begin
  foo = "INIT";
  $myTask();  // Call VPI function
  #0;
  $display("Final foo value: %s", foo);
end
```

The value of the foo string is initially set to "INIT" and then updated to "FAIL_TEST" through the VPI,  which is printed by the System verilog code as the final value.

 I've attached the test case I used for your reference. Feel free to modify it to show the tool issue you encountered. 
[issue-4082.tar.gz](https://github.com/user-attachments/files/18759713/issue-4082.tar.gz)

ktbarrett, 2025-02-12:
Removing the ifdef seems to work in 2024.3_2. Not sure when this was fixed, but we can close it.

ktbarrett, 2025-02-12:
Looks like it was 2024.3 that fixed this and CI is on 2024.2.

ktbarrett, 2025-02-12:
I'm not sure what happened, but running `nox` locally seemed to be fine, but it fails in CI. I'm glad their contrived example that uses DPI (we don't use DPI????) works, but it doesn't in more involved simulations, or non-DPI sims.

ktbarrett, 2025-02-12:
It could be something with these tests where the value is being inertially applied in the last read write phase of the time cycle. That's worth a look to help contextualize the bug. But that should still be legal if that is the case.

imphil, 2025-02-12:
Got back to Siemens to ask them if they can reproduce the issue with the cocotb test suite. At the same time it would be helpful if we can narrow the condition down that triggers the problem to a smaller test without full cocotb.

AbdouTharwat, 2026-03-02:
I tried running the test suite locally without the workaround and tests passed. I cannot reproduce this as well from my side

ktbarrett, 2026-03-03:
PR #5376 disables the hack.