cocotb/cocotb #4083: GLIBC_PRIVATE not found error with VCS U-2023.03

polyee13, 2024-07-31:
Attempting to run TinyALU example in pyuvm set up for VCS U-2023.03 and getting the following error. 
"/lib/x86_64-linux-gnu/libpthread.so.0: version `GLIBC_PRIVATE' not found (required by sim_build/simv)"

Has anyone come across this? I also created a case on Solvnet.
Cocotb version: 1.9.1
Ubuntu 22.04
VCS U-2023.3
Python 3.10.12 (self-installed)

```
acampos:~/gitspaces/pyuvm/examples/TinyALU (master)$ ode acampos:~/gitspaces/pyuvm/examples/TinyALU (master)$ make
rm -f results.xml
"make" -f Makefile results.xml
make[1]: Entering directory '/home/acampos/gitspaces/pyuvm/examples/TinyALU'
rm -f results.xml
MODULE=testbench TESTCASE= TOPLEVEL=tinyalu TOPLEVEL_LANG=verilog \
 sim_build/simv +define+COCOTB_SIM=1  -full64  
sim_build/simv: /lib/x86_64-linux-gnu/libpthread.so.0: version `GLIBC_PRIVATE' not found (required by sim_build/simv)
make[1]: *** [/home/acampos/.local/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.vcs:87: results.xml] Error 1
make[1]: Leaving directory '/home/acampos/gitspaces/pyuvm/examples/TinyALU'
make: *** [/home/acampos/.local/lib/python3.10/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
bash: syntax error near unexpected token `('
[1]+  Done                    code
rm -f results.xml
MODULE=testbench TESTCASE= TOPLEVEL=tinyalu TOPLEVEL_LANG=verilog \
 sim_build/simv +define+COCOTB_SIM=1  -full64  
sim_build/simv: /lib/x86_64-linux-gnu/libpthread.so.0: version `GLIBC_PRIVATE' not found (required by sim_build/simv)
make: *** [/home/acampos/.local/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.vcs:87: results.xml] Error 1
make[1]:: command not found
sim_build/simv: /lib/x86_64-linux-gnu/libpthread.so.0: version `GLIBC_PRIVATE' not found (required by sim_build/simv)

```

imphil, 2024-07-31:
The problem you're seeing is coming from VCS itself, not from cocotb.

"sim_build/simv: /lib/x86_64-linux-gnu/libpthread.so.0: version `GLIBC_PRIVATE' not found (required by sim_build/simv)"

So you'll need to get VCS running in your environment first before trying to use cocotb.

I'm not sure if VCS 2023.03 supports Ubuntu 22.04, please check with Synopsys.

polyee13, 2024-08-08:
I'm working with Synopsys on this. We run our sims that are outside of cocotb with no issues.

ktbarrett, 2024-08-08:
If regular sims are working, then the issue is probably that when cocotb's libraries are loaded, it is somehow loading system libraries before VCS can load its version of the same libraries with their custom ABI (`GLIBC_PRIVATE`) that its compiled against. You *could* `LD_PRELOAD` VCS's libraries or set `LD_LIBRARY_PATH` so their libraries get picked up over system libraries, which I'm surprised that VCS isn't already doing, but I have a sinking feeling that cocotb won't be able to use VCS's libraries because of the custom ABI.

The alternative is that you will need to compile cocotb from source using the customized GCC VCS probably ships with. This could be achieved by following the [development version installation instructions](https://docs.cocotb.org/en/stable/install_devel.html) after checking out the stable version of cocotb from Github while prepending the path to VCS's GCC binaries to your `PATH`.

I suggest following through with Synopsys on this. It seems they haven't considered their simulator needing to load externally-built libraries.

polyee13, 2024-08-09:
Let's close this issue. I seem to be past this problem and now have a different problem.

closed 2024-08-09 by polyee13

ktbarrett, 2024-08-09:
Could you provide some detail about your solution?

polyee13, 2024-08-12:
Honestly, I don't know how I got past it. It may have been related to not loading my tool env variables, but I couldn't duplicate the error anymore.