cocotb/cocotb #4088: verilator not generate dump.vcd file when has "EXTRA_ARGS += --trace --trace-structs" in Makefile

AnhHonag, 2024-08-07:


ktbarrett, 2024-08-07:
Add `--trace` to `PLUSARGS`. I agree something should be done to fix this, though I'm not entirely sure what the perfect solution is.

ktbarrett, 2024-08-09:
This was done in #3667 originally, and then backported to 1.9. I think the idea was that you could build for waves and then decide to collect waves or not when actually running the test, since with Verilator the hit is two-fold: compile time hit for compiling with trace support, and another bigger hit for actually collecting traces.

This works with the Python runner, but no thought was given to the Makefiles. That's something to fix for 1.9.1.

As for `cocotb-test`, downstream packages must be privy to upstream changes. Semver versions our public API, the verilator binaries we build are not part of our public API, so there is no guarantee. `cocotb-test` needs to update. Also, have you considered using cocotb's Python runner as it's very similar to, and in fact based upon, `cocotb-test`?

markusdd, 2024-08-09:
no we actually had just moved away from cocotbs runner as cocotb-test was better suited ;)

But I would argue against the API argument: the update broke all existing test setups that used dumping.
A backward compatible change would've been to say --no-trace to the simulator exucutable to suppress tracing even when it was built with it.

Usually you want to offer new features without breaking the existing setup.
And whats also particulary bad is that this came as a total surprise, we wasted like 4 hours trying to figure out why everything would compile with trace arguments and no dump.fst was being produced.

You can become picky about what is an API brick or not, but fact of the matter is not verilator changed it's argument on us, but the wrapper shipped by cocotb. I would disagree with not calling that our public API, also the necessary defines to enable dumping are part of this API.

I remember being very cautious to not break any existing vcd dumping when intriducing FST support back in 2020, this unfortunately wasn't done with this one.

I think there mere existence of this ticket by another user, and even me needing a few hours to figure out what's wrong, although I originally contributed FST tracing shows that this pretty mcuh should be considered public API ;)

At the minimum, if tracing is compiled in and --trace is not supplied there should be a log message stating that, because otherwise it silently breaks on a minor version upgrade and that's not cool.

imphil, 2024-08-09:
Let's please put the API break discussion to rest; Verilator support is [experimental](https://docs.cocotb.org/en/stable/simulator_support.html#verilator). Instead of discussing the past, let's see how we can move forward by doing actual work (as in: PRs and implementation-related discussions):

@markusdd, since you seem to have figured out what's missing: Are you interested in providing a PR to improve the situation? I'd suggest also providing a test to ensure we catch changes in behavior early.

Otherwise, someone else will pick up this item whenever their free time allows.

markusdd, 2024-08-09:
there is nothing missing really, the problem is we now already have a release out that changed API with the --trace option on the simulator executable. What should've been done is --no-trace for optional disable even if it's compiled in.
We could still do that, but then we break API yet AGAIN. So that ship has sailed.

cocotb-test allows to work around that by using the plus_args= argument to the run function. I read that cocotb Makefiles maybe cannot accomodate that but as I am not familiar with them/am not using them someone else needs to put in this option.

What I maybe could so is emit a warning if dump support is compile in but --trace is not provided to make the user aware he now needs to do this extra step which was not needed before.

PR #4089 (2024-08-14): Support SIM_ARGS in Verilator Makefile
Also, adds --trace to SIM_ARGS if VERILATOR_TRACE is set. Closes #4088. Backport to 1.9 once merged.
closed 2024-08-14 by ktbarrett