cocotb/cocotb #4099: Iterating through VPI objects with invalid vpi types

jakelongo, 2024-08-15:
Environment details:

- cocotb version tag v1.8.1
- OS Rocky 8.10 (64bit)
- VCS version V-2023.12-SP2-1 (behaviour is also present in T-2022.06-SP2-8) (64bit)
- Python 3.11 (virtual environment)

Traversing hierarchy through introspection or even iteration e.g. `print(dir(dut))` will trigger a VPI error in VCS (many in fact). 

I've raised a case with synopsys because I don't believe it should throw an error for attempting to get an iterator from a module for a valid vpi type (e.g. `vpiIntegerVar` or `vpiRealVar`). With that in mind, cocotb also tries to get an iterator for a `vpiRealNet` which seems to be a cadence only vpi definition (at least...that's my understanding from the comment).

Given the vpi libraries are already compiled per simulator, should these be placed in a header or gated behind a check in the iterator to avoid invalid vpi calls?

ktbarrett, 2024-08-15:
Thank you for opening a ticket with Synopsys for this.

> Given the vpi libraries are already compiled per simulator, should these be placed in a header or gated behind a check in the iterator to avoid invalid vpi calls?

Yes. The define is `IUS`. It should be an easy PR to throw up.

jakelongo, 2024-08-15:
I'm happy to have a go at the fix, though, I'm not completely sure what the regression test looks like as the valid iterate calls also cause problems in VCS. I'll have a closer look later this evening.

cmarqu, 2024-09-02:
> Yes. The define is `IUS`.

@ktbarrett Typo?

ktbarrett, 2024-09-03:
Yep. It's actually `VCS`.

jakelongo, 2024-09-05:
Apologies for not putting up a fix yet - the fix itself is easy but it wasn't obvious to me what the most appropriate way to to test it would be. Would you be happy with a PR that just prevents this req outside of IUS simulators?

I also think you were right the first time - it should be `IUS`. Cocotb shouldn't be trying to query that vpi type on anything besides Xcelium. 

> I've raised a case with synopsys because I don't believe it should throw an error for attempting to get an iterator from a module for a valid vpi type

☝️ This is still in progress. I've got them to fix another error which would set the exit code of the simulator incorrectly when this occurred but they are still struggling to understand this particular bug.

ktbarrett, 2024-09-05:
We have Xcelium running in our CI. If you have a fix, throw up a draft PR and see how the CI reacts.

We expect upstream issues to take a while, so no worries there.

imphil, 2025-01-29:
@jakelongo If you have it handy, can you post the Sysnopys case number here? That helps us and others when talking to Synopsys.

markusdd, 2025-02-24:
just found this issue. Is this possibly related to #4506 ?

Although I am getting the error messages with Xcelium, not VCS.

Also, for me, trying to use dict(dut) fails completely, set() or list() works though, but produces these ugly errors for vpiIntegerVar, vpiStructVar, vpiRealVar and vpiRealNet.

looks like this:
```
INFO     ERROR:          VPI       CANTIT                                                                                                                                                                                       
INFO              Can't iterate on type vpiRealNet from reference handle of type vpiModule.    
```