cocotb/cocotb #4108: how to write interface with dut has signal inout

AnhHonag, 2024-08-19:


imphil, 2024-08-19:
Hi, and welcome to cocotb. To be able to help you, please take a provide more information, as shown when you opened the issue:

> In case you are reporting a problem with running cocotb,
please remember to add relevant information about your environment, such as
> * the cocotb version used,
> * the operating system and version (32/64 bit),
> * the simulator and version (32/64 bit),
> * the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...),
> * (part of) the log file (cleaned of proprietary information).

In this case, we additionally need the SV/VHDL code, your testbench, and the error you're getting.

AnhHonag, 2024-08-26:
in systemverilog to dump wave file use bellow:
initial begin
// $display("input from test a=%b,b=%b,cin=%b",pif.a,pif.b,pif.cin);
$dumpfile("dump.vcd");
$dumpvars(1);
end

in cocotb, how will it be written?

closed 2024-08-26 by ktbarrett