cocotb/cocotb #4124: Cleanup exception types

ktbarrett, 2024-08-22:
* Is passing a negative number to `LogicArray.from_unsigned()` an OverflowError since it won't fit, or a ValueError, since the value is clearly not unsigned as required?
  * The fact some values of a given type yield ValueError and others OverflowError makes handling more difficult
* `Logic({})` yields `ValueError` rather than `TypeError`.

oscargus, 2024-09-30:
ValueError as "won't fit" really doesn't make sense. -1 fits in a single bit with two's complement/signed.

ktbarrett, 2025-02-17:
Is `Logic("0101")` an OverflowError or a ValueError? Is `Logic("1b8")` an OverflowError or a ValueError? Do we care enough about separating the types of exceptions to do the number of checks to make it more expressive?

Since ValueError has always been a possible exception type out of any of these objects and code that intends to handle exceptions should handle these, it should be very backwards compatible to just merge everything into `ValueError`. Separating all these errors into different types is only useful if the user would behave differently based on the failures. `TypeError` is useful because it signals something clearly wrong with the programmer's code, but all the invalid values are probably treated the same.

ktbarrett, 2025-02-21:
I figured out a backwards compatible solution. Make a new private Exception that inherits from both and we can use a `__subclass_check__` to give a deprecated warning for the OverflowError.

PR #4543 (2025-03-14): Use ValueError instead of OverflowError in sim objects
Closes #4124. Now there is no confusion on when to use `ValueError` or `OverflowError`. It's always `ValueError`.
closed 2025-03-14 by ktbarrett