cocotb/cocotb #4126: libgpilog broke on arm64 recently

paul-demo, 2024-08-23:
v1.8.1

This recently broke after working for 9 months, and I don't know why. Everything appears to be respecting the fact that this is an arm64 Mac, and yet the .so file that is generated for the libgpiolog file is evidently for an x86_64 architecture.

```bash
      running build_ext
      building 'cocotb/libs/libgpilog' extension
      creating build/temp.macosx-13.6-arm64-cpython-39
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog/cocotb
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog/cocotb/share
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog/cocotb/share/lib
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log
      clang++ -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DOPENSSL_NO_SSL3 -DGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -Icocotb/share/include -Icocotb -I/obfuscated_dir/.venv/include -I/obfuscated_user_dir/.asdf/installs/python/3.9.17/include/python3.9 -c cocotb/share/lib/gpi_log/gpi_logging.cpp -o build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wno-missing-field-initializers -Wnon-virtual-dtor -Woverloaded-virtual -flto
      creating build/lib.macosx-13.6-arm64-cpython-39/cocotb/libs
      clang++ -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DOPENSSL_NO_SSL3 -bundle -undefined dynamic_lookup build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o -Lbuild/lib.macosx-13.6-arm64-cpython-39/cocotb/libs -L/obfuscated_user_dir/.asdf/installs/python/3.9.17/lib -o build/lib.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog.so -flto -Wl,-install_name,@rpath/libgpilog.so -Wl,-rpath,@loader_path
      building 'cocotb/libs/libpygpilog' extension
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog/cocotb
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog/cocotb/share
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog/cocotb/share/lib
      creating build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog/cocotb/share/lib/py_gpi_log
      clang++ -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DOPENSSL_NO_SSL3 -DPYGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -Icocotb/share/include -Icocotb -I/obfuscated_dir/.venv/include -I/obfuscated_user_dir/.asdf/installs/python/3.9.17/include/python3.9 -c cocotb/share/lib/py_gpi_log/py_gpi_logging.cpp -o build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog/cocotb/share/lib/py_gpi_log/py_gpi_logging.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wno-missing-field-initializers -Wnon-virtual-dtor -Woverloaded-virtual -flto
      clang++ -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -DOPENSSL_NO_SSL3 -bundle -undefined dynamic_lookup build/temp.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog/cocotb/share/lib/py_gpi_log/py_gpi_logging.o -Lbuild/lib.macosx-13.6-arm64-cpython-39/cocotb/libs -L/obfuscated_user_dir/.asdf/installs/python/3.9.17/lib -lgpilog -o build/lib.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog.so -flto -Wl,-install_name,@rpath/libpygpilog.so -Wl,-rpath,@loader_path
      ld: unsupported mach-o filetype (only MH_OBJECT and MH_DYLIB can be linked) in '/private/var/folders/xs/wypsst212vqc_4xl1kf59c7r0000gr/T/pip-install-2sj_iw8v/cocotb_5c8005f5d0174b59a5599d9f9a2af182/build/lib.macosx-13.6-arm64-cpython-39/cocotb/libs/libgpilog.so'
      clang: error: linker command failed with exit code 1 (use -v to see invocation)
      error: command '/usr/bin/clang++' failed with exit code 1

```

paul-demo, 2024-08-23:
That last `clang++` command should be creating the .so file due to `-o build/lib.macosx-13.6-arm64-cpython-39/cocotb/libs/libpygpilog.so` (if you scroll to the right).

For some reason, even though this is not running in x86_64/Rosetta (the parent script is started with `arch -arm64`), the clang++ operation is apparently producing an x86_64 binary...

paul-demo, 2024-08-23:
Could this be caused by this recent change where you started overriding CFLAGS?

https://github.com/cocotb/cocotb/commit/1929837457788935c6c1f87cb3bc7c0108e442ff#diff-f7a16a65f061822bcc73b8296f4dc837353d379d8d9cc5307982cb6941442835R123

ktbarrett, 2024-08-23:
This was due to a change to `setuptools` that broke us. https://github.com/cocotb/cocotb/issues/4094. This has been fixed on master as well as the stable/1.9 branch, which should be compatible with v1.8.1.

closed 2024-08-23 by ktbarrett

paul-demo, 2024-08-23:
Wonderful. Thanks for the quick reply. This timeline (Aug 13th ish) aligns exactly with when my system broke.

paul-demo, 2024-08-23:
Do you intend to publish the updated 1.9.0 branch with those fixes to pypi?

ktbarrett, 2024-08-25:
Yes, eventually.

robtaylor, 2024-09-05:
Any chance sometime soon? This is breaking quite a few things for me at the moment... or maybe backport to a 1.8.x?

ktbarrett, 2024-09-05:
This has already been fixed and released in 1.9.1.