cocotb/cocotb #4150: GHDL: Deposit does not work as expected. It forces the signal

sacatak, 2024-09-06:
Hello,
my cocotb-version:1.8.1.
Deposit works as a force command on GHDL. In addition, it is impossible to release it.
For example with this vhdl code and cocotb test:
```
process (clk, resetn)
begin
    if resetn = '0' then
        no_used_signal           <= '0';
    elsif rising_edge(clk) then
        no_used_signal           <= not(no_used_signal);
    end if;
end process;
```

```
@cocotb.test()
async def test_nothing(dut):
    await config_general(dut)

    await Timer(100, "ns")
    await RisingEdge(dut.no_used_signal)
    dut.no_used_signal.value = Deposit(0)
    
    await Timer(100, "ns")
```

Once the signal is deposit to 0 under GHDL, il never updates again:
[image]
Under Questa it works:
[image]

ktbarrett, 2024-09-06:
Some investigation should be done to ensure what we think is happening is actually happening, but this does seem like an upstream bug, so I will mark it as such until we are certain it isn't.

sebsky-dev, 2024-11-04:
@tgingold I'm also having the issue where the deposit on cocotb behaves as a force. Any idea where to look ? How can I debug this issue ?

tgingold, 2024-11-05:
@sebsky-dev It would be nice to have a reproducer...

svancau, 2024-11-05:
I cannot publish the problematic code as is. I'll write you the python/vhdl package to show this :)

sebsky-dev, 2024-11-05:
I cannot public problematic code as is. I'll write you a VHDL/Python example :)

sebsky-dev, 2024-11-05:
@tgingold I can reproduce this bug the same way as above.
This test has been done on a Fedora 41, packaged GHDL v3.0.0 with cocotb v1.9.2

[image]
With GHDL, @69ns, o should be 01, it is not driven by the VHDL design. Fails

[image]
With Modelsim, as expected, when output is enabled, the d is copied on the output. Passes
[test_deposit_bug.zip](https://github.com/user-attachments/files/17636362/test_deposit_bug.zip)

tgingold, 2024-11-06:
Thank you, I can reproduce the issue.

sebsky-dev, 2024-11-07:
Do you have any timeline for the fix ?

tgingold, 2024-11-08:
I have started to work on it.

tgingold, 2024-11-10:
@sebsky-dev, this is now fixed in ghdl.  Can you confirm ?

sebsky-dev, 2024-11-12:
Yes, it's working, thanks for your fast fix :)

closed 2024-11-12 by ktbarrett