cocotb/cocotb #4154: Problem running cocotb with Questa and Modelsim

TarikHamedovic, 2024-09-11:
I am trying to run a Lattice IP simulation using Questa or Modelsim. This problem that I will explain will be for Questa, but the same problem is present in Modelsim.
Here is the terminal output after running just a simple cocotb testbench for a Multiplier IP.
```
make test SIMULATOR=questa TOPLEVEL=top MODULE=multiplier_tb
make[1]: Entering directory '/home/user/SDR-HLS/1.RTLImplementation/2.sim/Multiplier'
make -C . SIM=questa TOPLEVEL_LANG=verilog TOPLEVEL=top MODULE=multiplier_tb
make[2]: Entering directory '/home/user/SDR-HLS/1.RTLImplementation/2.sim/Multiplier'
rm -f results.xml
make -f Makefile results.xml
make[3]: Entering directory '/home/user/SDR-HLS/1.RTLImplementation/2.sim/Multiplier'
mkdir -p sim_build
# Make sure all libs in SOURCES_VHDL_* are mentioned in VHDL_LIB_ORDER and vice versa
rm -f results.xml
set -o pipefail; MODULE=multiplier_tb TESTCASE= TOPLEVEL=top \
GPI_EXTRA= TOPLEVEL_LANG=verilog \
 /home/user/FPGA/tools/radiant/questasim/bin/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading pref.tcl

# 2024.2

# do sim_build/runsim.do
# Questa Lattice OEM Edition-64 vmap 2024.2 Lib Mapping Utility 2024.05 May 20 2024
# vmap -c 
# Copying /home/user/FPGA/tools/radiant/questasim/linux_x86_64/../modelsim.ini to modelsim.ini
# Questa Lattice OEM Edition-64 vmap 2024.2 Lib Mapping Utility 2024.05 May 20 2024
# vmap work sim_build/work 
# Modifying modelsim.ini
# vsim -onfinish exit -pli "/home/user/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbvpi_modelsim.so" sim_build/top 
# Start time: 15:24:36 on Sep 11,2024
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vsim-3865) Due to PLI being present, global +acc is being enabled automatically. This will cause your simulation to run very slowly. Please use vsim -no_autoacc to disable this feature. This feature is now deprecated and will be removed from future releases.
# ** Warning: (vopt-10587) Some optimizations are turned off because the +acc switch is in effect. This will cause your simulation to run slowly. Please use -access/-debug to maintain needed visibility. The +acc switch would be deprecated in a future release.
# ** Error: (vopt-11) Could not find work.sim_build/top.
# Optimization failed
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=1, Warnings=2.
# Error loading design
Error loading design
# End time: 15:24:37 on Sep 11,2024, Elapsed time: 0:00:01
# Errors: 1, Warnings: 2
make[3]: *** [/home/user/.local/lib/python3.10/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:160: results.xml] Error 12
make[3]: Leaving directory '/home/user/SDR-HLS/1.RTLImplementation/2.sim/Multiplier'
make[2]: *** [/home/user/.local/lib/python3.10/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
make[2]: Leaving directory '/home/user/SDR-HLS/1.RTLImplementation/2.sim/Multiplier'
make[1]: *** [Makefile:39: test] Error 2
make[1]: Leaving directory '/home/user/SDR-HLS/1.RTLImplementation/2.sim/Multiplier'
make: *** [Makefile:44: run] Error 2

```
The main problem is detailed in this line
```
# ** Error: (vopt-11) Could not find work.sim_build/top.
```

Versions:
cocotb: Version: 1.8.1
OS: Ubuntu 22.04 LTS x86_64
Simulator: Questa Lattice OEM Edition-64 vsim 2024.2 Simulator 2024.05 May 20 2024
Python: Python 3.10.12

ktbarrett, 2024-09-14:
Can you provide the Makefile you are using for us to debug?

TarikHamedovic, 2024-09-14:
Hello, 

You can find the example [here](https://github.com/TarikHamedovic/SDR-HLS/tree/main/1.RTLImplementation/2.sim/Multiplier). Changing the simulator to `modelsim` or `questa` gives back the same error.

I run the simulation using `make run`.

toitoise, 2024-09-20:
Hi,
Got same issue and fixed with this (Issue in Makefile for Questa (cocotb/share/makefiles/simulators/Makefile.questa))  :
>       #@echo "vsim $(VSIM_ARGS) $(EXTRA_ARGS) $(PLUSARGS) $(SIM_BUILD)/$(TOPLEVEL)" >> $@
replaced by:
>       @echo "vsim $(VSIM_ARGS) $(EXTRA_ARGS) $(PLUSARGS) **$(TOPLEVEL)**" >> $@

Just have to give name of the toplevel not prefixed by the path to.

ktbarrett, 2024-09-20:
This makes sense, but I don't understand why it was working beforehand.

PR #4190 (2024-10-09): Fix toplevel discovery in Questa and Modelsim
Closes #4154.
closed 2024-10-09 by ktbarrett