cocotb/cocotb #4159: Allow bool casts of single-bit LogicArray instances

imphil, 2024-09-18:
Imagine code like this:

```
if not dut.fifo.empty.value:
  pass
```

This code will now raise a `FutureWarning` about bool casts not being supported on LogicArray. For single-bit values, the bool cast is defined without much risk for confusion (`LogicArray("1") == true`, everything else is false). Can we make this special case supported?

ktbarrett, 2024-09-19:
Is the impetus for this to compare against the values of scalar logic handles? The intent here in the future is to separate LogicObject and LogicArrayObject and return a Logic object as a value from the scalar version. The scalar Logic type is already castable to bool.

imphil, 2024-09-19:
> Is the impetus for this to compare against the values of scalar logic handles?

Yes.

Valid/ready/full/empty signals are something that's commonly used as bool, and getting a warning whenever I do `if not dut.full.value` seems excessive.

Do you have more specific plans for when to return LogicObject for scalar handles? I.e., is "the future" "within the cocotb 2.0 timeframe"?

ktbarrett, 2024-09-19:
I was hoping. Now that it's shifted to "Holidays 2024" I have some more time to actually implement this change.

ktbarrett, 2025-01-12:
#4318 separates `LogicObject` into scalar `LogicObject` and array'd `LogicArrayObject`, which return scalar `Logic` and `LogicArray` respectively. `Logic.__bool__` works as one would expect, #4143 introduced a deprecated bool cast for LogicArray which mimics `BinaryValue`'s behavior.

closed 2025-01-12 by ktbarrett