cocotb/cocotb #4161: How to force x/z states to a signal？

Elefseus, 2024-09-20:
hello, my cocotb version is 1.9.1, and I want to force signal with some 'x' or 'z' value, such as
```python
xxx.value = Force('xxxzzz')
```
but I found it didn't support,  and I have tried with
```python
xxx.value = Realse()
xxx.value.assign('xxxzzz')
xxx.value = Freeze()
```
but it didn't work as I expected.
In cocotb, how to force x/z states to a signal？

Elefseus, 2024-09-20:
> This should be supported. Please add details to your question about your simulator and simulator version.

my simulator is vcs 2018.09

when I do force, it shows that cannot Force str to a BinaryValue:
```python
xxx.value = Force('1001xxzz')
TypeError: Unsupported type for value assignment:<class'str'> ('1001xxzz')
```

now I find a way to do this, and it works:
```python
xxx.value = Force(LogicArray('1001xxzz'.zfill(len(xxx.value))
```
is there a better way ?

ktbarrett, 2024-09-20:
In cocotb 1.9 no. Handles do not support being assigned from string directly. In 2.0 they will be though. As for the `zfill` section, that will not be changed in 2.0. All values must be of the same size as the handle they are modifying. cocotb will not guess what you intend with the unset bits.

closed 2024-09-23 by ktbarrett