cocotb/cocotb #4164: Python Runner with Xcelium sets value for all parameters in the design with the same name

Ustin, 2024-09-20:
Cocotb 1.9.1.

I'm trying to set up a Verilog parameters in a testbench using Python runner. 

When using Python runner with Xcelium, it uses -gpg "object_name => value", which according to the documentation "assigns a value to all generics and parameters in the design using a specified name."
This breaks compatibility of testbenches with Verilator. For verilator it uses -G\<name\>=\<value\>, which according to the [documentation](https://veripool.org/guide/latest/exe_verilator.html) sets parameters only on top level.

According to Cadence docs alternative would be using -generic option to assign values to VHDL generics and -defparam option for verilog parameters.

Ustin, 2024-09-20:
Temporary solution I found with some monkey patching: 

Before creating python runner:
```
  ...
  hdl_toplevel = "lms_nco_freq_converter"

  @staticmethod
  def _get_parameter_options(parameters: Mapping[str, object]) -> cocotb.runner.Command:
    return [f'-defparam {hdl_toplevel}.{name}={value}' for name, value in parameters.items()]

  cocotb.runner.Xcelium._get_parameter_options = _get_parameter_options

  runner = get_runner("xcelium")
  runner.build( ...
```

Works for Verilog, slightly different with VHDL.

ktbarrett, 2024-10-01:
Also we should perhaps support top-level vs hierarchical parameters as separate options so any given options works the same.

ktbarrett, 2024-12-02:
We fixed this internally by prefixing the name of the param with the top module followed by a colon. i.e. `{top_module}:{name} => {value}`.