cocotb/cocotb #4171: JUnit output is not valid

ktbarrett, 2024-09-23:
### Discussed in https://github.com/cocotb/cocotb/discussions/4165

<div type='discussions-op-text'>

<sup>Originally posted by **michael-miftek** September 20, 2024</sup>
I am running into some issues with reading the Cocotb xml file with XUnit and was wondering if there were any suggestions or if anyone had run into similar issues and how they fixed it.

I have installed the `XUnit` plugin on Jenkins and am able to run `Cocotb` (at this moment I am just using the demo files form the Quickstart Guide), the results populate to `{root}/sim_build/results.xml` and I use the Post-Build Actions `Publish xUnit test result report` with `JUnit` test and point it to `sim_build/*.xml`. The results file is found but then this error is output (shown below), connecting this to Jenkins has been a pain thus far and was wondering if anyone had any points of help, I am just trying to setup a basic freestyle project for now to see how things work between cocotb and Jenkins (using xUnit but open to other options).

When running just the command `junit 'sim_build/*.xml'` it registers the results and if a test is failed it notes it as expected was just hoping to use xUnit to have more granularity on test control and wondering if anyone here was able to offer more help?

```INFO: [JUnit] - 1 test report file(s) were found with the pattern 'sim_build/*.xml' relative to 'C:\Jenkins\Hardware\workspace\firmwaretest' for the testing framework 'JUnit'.
WARNING: The file 'C:\Jenkins\Hardware\workspace\firmwaretest\sim_build\results.xml' is an invalid file.
WARNING: At line 2 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.4: Attribute 'tests' must appear on element 'testsuite'.
WARNING: At line 2 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.4: Attribute 'failures' must appear on element 'testsuite'.
WARNING: At line 2 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.4: Attribute 'errors' must appear on element 'testsuite'.
WARNING: At line 3 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.2.4.a: Invalid content was found starting with element 'property'. One of '{testsuite, properties, testcase, system-out, system-err}' is expected.
WARNING: At line 4 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'file' is not allowed to appear in element 'testcase'.
WARNING: At line 4 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'lineno' is not allowed to appear in element 'testcase'.
WARNING: At line 4 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'sim_time_ns' is not allowed to appear in element 'testcase'.
WARNING: At line 4 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'ratio_time' is not allowed to appear in element 'testcase'.
WARNING: At line 5 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'file' is not allowed to appear in element 'testcase'.
WARNING: At line 5 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'lineno' is not allowed to appear in element 'testcase'.
WARNING: At line 5 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'sim_time_ns' is not allowed to appear in element 'testcase'.
WARNING: At line 5 of file:/C:/Jenkins/Hardware/workspace/firmwaretest/sim_build/results.xml:cvc-complex-type.3.2.2: Attribute 'ratio_time' is not allowed to appear in element 'testcase'.
ERROR: Step ‘Publish xUnit test result report’ failed: The result file 'C:\Jenkins\Hardware\workspace\firmwaretest\sim_build\results.xml' for the metric 'JUnit' is not valid. The result file has been skipped.```</div>

imphil, 2024-09-23:
Looks like there are at least two XML Schema files we could use:
* https://github.com/windyroad/JUnit-Schema/blob/master/JUnit.xsd
* https://github.com/junit-team/junit5/blob/main/platform-tests/src/test/resources/jenkins-junit.xsd

I have yet to find the "official documentation" for the format (and I doubt it exists). The comment "Note that the generated XML format is compatible with the legacy de facto standard for JUnit 4 based test reports that was made popular by the Ant build system." from the [Junit5 source code](https://junit.org/junit5/docs/current/api/org.junit.platform.reporting/org/junit/platform/reporting/legacy/xml/LegacyXmlReportGeneratingListener.html) indicates that actually Apache Ant defined the format, not Junit itself? https://github.com/ota4j-team/open-test-reporting (Prior Art) goes into the same direction. 

Instead of fixing a only vaguely defined, how about we keep what cocotb currently has to keep existing users happy, and additionally provide an alternative, well-defined format? Maybe https://github.com/ota4j-team/open-test-reporting is one of those options, but I'm sure there are other, similar efforts as well.

oscargus, 2024-09-30:
VUnit has support for two different XML-formats (JUnit like) and the documentation says:
```
“jenkins” = Output stored in <system-out>, “bamboo” = Output stored in <failure>
```