cocotb/cocotb #4175: Questa runner _build_command creates new command for each vhdl_sources preventing the use of autoorder.

klemen-erjavec-bl, 2024-09-25:
When I define vhdl_sources for runner.build this generates vcom command for each vhdl source, this is different as it was before when one vcom command was created for all the vhdl_sources instead.

```python

    vhdl_sources = [
        *glob('../some_folder/*.vhd'),
        *glob('../some_other_folder/*.vhd')
    ]
    runner.build(
        vhdl_sources = vhdl_sources,
        build_args = ['-autoorder']
    )

```

before change made in:
https://github.com/cocotb/cocotb/commit/e786e02a935758e2712c17067bc5a130bd1f7163#diff-e1299e8e859a25007d723b0daee6c4f1d9277ff2712d01be145d0550dd9a7148R726
see src/cocotb_tools/runner.py line 634
https://github.com/cocotb/cocotb/pull/3796

This would create vcom which received all vhd files and used **autoorder** to sort dependency of vhd files.
```
vcom -work top -autoorder path_vhd1.vhd path_vhd2.vhd path_vhd3.vhd
```

With the mentioned change vcom now generates three commands instead of one. vcom then fails because path_vhd1.vhd depends on path_vhd3.vhd and autoorder cannot be used.

```
vcom -work top -autoorder path_vhd1.vhd
vcom -work top -autoorder path_vhd2.vhd 
vcom -work top -autoorder path_vhd3.vhd
```

ktbarrett, 2024-09-30:
We need to support language-specific build args as well, which requires separate compilation steps. I suggest creating your own test runner implementation by inheriting from the `Runner` class to support this flow.

The Python runner implementation we have is intended to provide a single implementation to get something to *work*. We are not intending what we provide to be maximally efficient or capable. Users can extend `Runner`, consult [the documentation on using cocotb in an existing flow](https://docs.cocotb.org/en/stable/custom_flows.html), or perhaps use some other tool whose goal is to be project automation, like [edalize](https://github.com/olofk/edalize).

ktbarrett, 2024-10-01:
Alternatively, using `autoorder` could be standard if it's been supported for a reasonably long time, say 4+ years?

ktbarrett, 2024-10-01:
> Unless cocotb is able to do that on its own and I am maybe now aware of it?

No. As you've probably seen while editing the code, the runner is pretty basic. 

Glad you found something that works for you!

closed 2024-10-01 by ktbarrett