cocotb/cocotb #4201: GHDL master fails to access signals?

imphil, 2024-10-07:
GHDL master in our experimental tool CI bucket seems to fail on a wide range of tests, e.g.:

```
make[4]: Entering directory '/home/runner/work/cocotb/cocotb/examples/simple_dff'
mkdir -p sim_build
\
/usr/local/bin/ghdl -i   --workdir=sim_build --work=work /home/runner/work/cocotb/cocotb/examples/simple_dff/dff.vhdl && \
/usr/local/bin/ghdl -m   --workdir=sim_build -Psim_build --work=work dff 
/home/runner/work/cocotb/cocotb/.nox/dev_test_sim-sim-ghdl-toplevel_lang-vhdl-gpi_interface-vpi/lib/python3.8/site-packages/cocotb_tools/makefiles/simulators/Makefile.ghdl:105: Using MODULE is deprecated, please use COCOTB_TEST_MODULES instead.
rm -f results.xml
COCOTB_TEST_MODULES=test_dff COCOTB_TESTCASE= COCOTB_TEST_FILTER= COCOTB_TOPLEVEL=dff TOPLEVEL_LANG=vhdl \
 /usr/local/bin/ghdl -r  --time-resolution=ps --workdir=sim_build -Psim_build --work=work dff  --vpi=/home/runner/work/cocotb/cocotb/.nox/dev_test_sim-sim-ghdl-toplevel_lang-vhdl-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so   
loading VPI module '/home/runner/work/cocotb/cocotb/.nox/dev_test_sim-sim-ghdl-toplevel_lang-vhdl-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/runner/work/cocotb/cocotb/.nox/dev_test_sim-sim-ghdl-toplevel_lang-vhdl-gpi_interface-vpi/bin/python
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     cocotb                             Running on GHDL version 5.0.0-dev (4.1.0.r253.geeab69c29) [Dunoon edition]
     0.00ns INFO     cocotb                             Running tests with cocotb v2.0.0.dev0+e8c6123 from /home/runner/work/cocotb/cocotb/.nox/dev_test_sim-sim-ghdl-toplevel_lang-vhdl-gpi_interface-vpi/lib/python3.8/site-packages/cocotb
     0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
     0.00ns INFO     cocotb                             Seeding Python random module with 1728269520
     0.00ns INFO     cocotb.regression                  running test_dff.dff_simple_test (1/1)
                                                            Test that d propagates to q
     0.00ns INFO     cocotb.regression                  test_dff.dff_simple_test failed
                                                        Traceback (most recent call last):
                                                          File "/home/runner/work/cocotb/cocotb/examples/simple_dff/test_dff.py", line 31, in dff_simple_test
                                                            assert LogicArray(dut.q.value) == initial
                                                          File "/home/runner/work/cocotb/cocotb/.nox/dev_test_sim-sim-ghdl-toplevel_lang-vhdl-gpi_interface-vpi/lib/python3.8/site-packages/cocotb/handle.py", line 442, in __getattr__
                                                            raise AttributeError(str(e)) from None
                                                        AttributeError: 'dff.q contains no child object named value'
     0.00ns INFO     cocotb.regression                  **************************************************************************************
                                                        ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
                                                        **************************************************************************************
                                                        ** test_dff.dff_simple_test       FAIL           0.00           0.00          0.00  **
                                                        **************************************************************************************
                                                        ** TESTS=1 PASS=0 FAIL=1 SKIP=0                  0.00           0.01          0.00  **
                                                        **************************************************************************************
```

* See https://github.com/cocotb/cocotb/actions/runs/11207759663/job/31150428483
* GHDL geeab69c29

ktbarrett, 2024-10-07:
Is GHDL returning a different VPI type for std_logic signals now?

imphil, 2024-10-10:
@tgingold Do you have an idea what this problem came from? Looks like there were recent changes to the VPI interface in GHDL (https://github.com/ghdl/ghdl/commits/master/src/grt/grt-vpi.adb).

tgingold, 2024-10-11:
The last main change to the VPI interface was support of vpiVectVal.
Can you run with `--vpi-trace` to get more data about the failure ?