cocotb/cocotb #4202: Verilator: test_cases/test_plusargs fails verilation with compile error on MacOS

imphil, 2024-10-07:
* MacOS 13.7 22H123
* Actions run: https://github.com/cocotb/cocotb/actions/runs/11207759663/job/31150429093

```
g++ -std=gnu++11 -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=1 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -std=gnu++17 -fcoroutines-ts -c -o verilated_timing.o /usr/local/share/verilator/include/verilated_timing.cpp
In file included from /usr/local/share/verilator/include/verilated_timing.cpp:23:
/usr/local/share/verilator/include/verilated_timing.h:104:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:118:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:208:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:276:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:324:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:374:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) const { coro.destroy(); }
                       ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:409:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:426:14: error: no template named 'coroutine_handle' in namespace 'std'
        std::coroutine_handle<> m_continuation;  // Coroutine to resume after this one finishes
        ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:434:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never initial_suspend() const { return {}; }
        ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:438:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never final_suspend() noexcept;
        ~~~~~^
/usr/local/share/verilator/include/verilated_timing.h:473:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) { m_promisep->m_continuation = coro; }
In file included from /usr/local/share/verilator/include/verilated_timing.h:28:
In file included from /usr/local/share/verilator/include/verilated.h:59:
In file included from /Applications/Xcode_15.2.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/map:544:
/Applications/Xcode_15.2.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__tree:2264:26: error: static_cast from '__node_pointer' (aka 'std::__tree_node<std::__value_type<unsigned long long, VlCoroutineHandle>, void *> *') to '__node_base_pointer' (aka 'std::__tree_node_base<void *> *'), which are not related by inheritance, is not allowed
                         static_cast<__node_base_pointer>(__ptr));
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
16 errors generated.
```

@ktbarrett Is this one of the bugs you reported upstream that's still outstanding? Let's link them here to have a better overview of what's missing in our experimental tool version bucket.

ktbarrett, 2024-10-07:
This is an upstream issue. We thought it was fixed, but it is not. It seems to be due to a bug in autoconf that has been fixed upstream there, but not rolled out yet.

mornthx, 2025-07-31:
I have a similar fail case on Ubuntu 2022.4, cocotb 1.9.2 and verilator 5.036.

**/usr/include/c++/11/coroutine:334:2: error: #error "the coroutine header requires -fcoroutines"**
and there are a lot of error following as shown below.

I wonder if it's the same issue reported there? and if it is, is it possible to fix the bug manually?

`In file included from /usr/local/include/verilated_timing.h:51,
from /usr/local/include/verilated_timing.cpp:23:
/usr/include/c++/11/coroutine:334:2: error: #error "the coroutine header requires -fcoroutines"
  334 | #error "the coroutine header requires -fcoroutines"
      |  ^~~~~
In file included from /usr/local/include/verilated_timing.cpp:23:
/usr/local/include/verilated_timing.h:104:10: error: ‘coroutine_handle’ in namespace ‘std’ does not name a template type
  104 |     std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
      |          ^~~~~~~~~~~~~~~~
/usr/local/include/verilated_timing.h:118:44: error: expected ‘)’ before ‘<’ token
  118 |     VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
      |                      ~                     ^
      |                                            )`

imphil, 2025-07-31:
This is an upstream verilator issue. Please report the issue there. You might want to upgrade to the latest Verilator release before.

mornthx, 2025-08-01:
Sure. I have solved this issue, I'll open another issue.