cocotb/cocotb #4222: Reading a synchronous signal on clock rising edge will return the previous value

Cepadues, 2024-10-16:
Hello,

I'm running on some issues with the value of signals not being updated fast enough.
On a demo project, I have a simple "logic-and" vhdl entity that outputs '1' when both inputs are set to '1'.
The output is synchronous of the clock signal, so in my testbench I have :

```python
dut.input_1 = 1
dut.input_2 = 1

await RisingEdge(dut.clock)

assert dut.output = 1
```

Which will cause a failure as the returned value of dut.output is '0' while the waves show that it has been driven high on this clock rising edge.

This has also been the case with ModelsimDE.

My environment :
* Cocotb version 1.9.1,
* Windows 11 64 bits,
* Questasim 2022.3 64 bits,
* Python 3.12.6 64 bits,

ktbarrett, 2024-10-16:
This is expected behavior in an event-based simulator. Just like HDL when you have a process that's sensitive to `rising_edge(clk)`, you read the stable value at the end of the last cycle. cocotb is like HDL, not like waveforms. Waveforms are an anachronism.

If you want to get the final quiesced values in a given stage use `await ReadOnly()`.

closed 2024-10-16 by ktbarrett

Cepadues, 2024-10-16:
The function you suggested is just what I needed.
I understand the mechanism better now, it's true that there's no way for the simulator to know what action cocotb will perform, unlike in a vhdl testbench.
I've added the `await ReadOnly()` to my check function.
Thank you