cocotb/cocotb #4238: test_custom_entry test is failing

vinomutty, 2024-10-28:
we cloned the below and tried running the testcases with make SIM=questa TOPLEVEL_LANG=vhdl
git clone --branch stable/1.9 --single-branch https://github.com/cocotb/cocotb

We tried to run test_custom_entry test which is failing with below error message

ERROR: results.xml was not written by the simulation!
make[2]: *** [/home/hemashri.bhagavati/.local/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:161: results.xml] Error 1
make[2]: Leaving directory '/Projects/marmik_project/hemashri.bhagavati/coco_new/cocotb/tests/test_cases/test_custom_entry'
make[1]: *** [/home/hemashri.bhagavati/.local/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
make[1]: Leaving directory '/Projects/marmik_project/hemashri.bhagavati/coco_new/cocotb/tests/test_cases/test_custom_entry'
make: [Makefile:6: override_for_this_test] Error 2 (ignored)

imphil, 2024-10-28:
That's not a failure, that's expected. make should return a zero return code.

vinomutty, 2024-10-28:
Please can you elaborate "make should return a zero return code"?

vinomutty, 2024-10-28:
1.Test_exit_error
2.Test_select_testcase_error
Above testcases as name suggest should be failing.Right?

vinomutty, 2024-10-29:
> That's not a failure, that's expected. make should return a zero return code.

wavedump not showing signals. is it expected?

[image]

ktbarrett, 2024-10-29:
This test exists only to test the PYGPI_ENTRY_POINT variable. It passes no time, and there is no results.xml, so it fails the Makefile's default test failure check, but it is checked another way if you look at the Makefile.

closed 2024-10-29 by ktbarrett