cocotb/cocotb #4245: NVC DLL missing on Windows builds

Gophen, 2024-10-30:
Hello,

I've been trying to run NVC with cocotb on Windows but so far without success.
The VHPI library seems to be missing from local build (master branch) and from [precompiled windows python binaries](https://pypi.org/project/cocotb/#files) but not from Linux.

Logs :
>** Note: loading VHPI plugin .venv/Lib/site-packages/cocotb/libs/cocotbvhpi_nvc.dll
>** Fatal: failed to load .venv\Lib\site-packages\cocotb\libs\cocotbvhpi_nvc.dll: DLL file is missing

Files in  .venv\Lib\site-packages\cocotb\libs:
```bash
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a----        30/10/2024     11:17          26112 cocotb.dll
-a----        30/10/2024     11:17           1109 cocotb.exp
-a----        30/10/2024     11:17           2358 cocotb.lib
-a----        30/10/2024     11:17          86016 cocotbfli_modelsim.dll
-a----        30/10/2024     11:17            875 cocotbfli_modelsim.exp
-a----        30/10/2024     11:17           2176 cocotbfli_modelsim.lib
-a----        30/10/2024     11:17          11776 cocotbutils.dll
-a----        30/10/2024     11:17            974 cocotbutils.exp
-a----        30/10/2024     11:17           2146 cocotbutils.lib
-a----        30/10/2024     11:17          78848 cocotbvhpi_aldec.dll
-a----        30/10/2024     11:17           1073 cocotbvhpi_aldec.exp
-a----        30/10/2024     11:17           2442 cocotbvhpi_aldec.lib
-a----        30/10/2024     11:17          78848 cocotbvhpi_modelsim.dll
-a----        30/10/2024     11:17           1080 cocotbvhpi_modelsim.exp
-a----        30/10/2024     11:17           2484 cocotbvhpi_modelsim.lib
-a----        30/10/2024     11:17          66560 cocotbvpi_aldec.dll
-a----        30/10/2024     11:17           1068 cocotbvpi_aldec.exp
-a----        30/10/2024     11:17           2418 cocotbvpi_aldec.lib
-a----        30/10/2024     11:17          66560 cocotbvpi_ghdl.dll
-a----        30/10/2024     11:17           1067 cocotbvpi_ghdl.exp
-a----        30/10/2024     11:17           2408 cocotbvpi_ghdl.lib
-a----        30/10/2024     11:17           1071 cocotbvpi_icarus.exp
-a----        30/10/2024     11:17           2438 cocotbvpi_icarus.lib
-a----        30/10/2024     11:17          66048 cocotbvpi_icarus.vpl
-a----        30/10/2024     11:17          66048 cocotbvpi_modelsim.dll
-a----        30/10/2024     11:17           1075 cocotbvpi_modelsim.exp
-a----        30/10/2024     11:17           2468 cocotbvpi_modelsim.lib
-a----        30/10/2024     11:17          12288 embed.dll
-a----        30/10/2024     11:17           1096 embed.exp
-a----        30/10/2024     11:17           2322 embed.lib
-a----        30/10/2024     11:17          47104 gpi.dll
-a----        30/10/2024     11:17          21031 gpi.exp
-a----        30/10/2024     11:17          34400 gpi.lib
-a----        30/10/2024     11:17          15872 gpilog.dll
-a----        30/10/2024     11:17           1632 gpilog.exp
-a----        30/10/2024     11:17           3244 gpilog.lib
-a----        30/10/2024     11:17          17408 pygpilog.dll
-a----        30/10/2024     11:17           1043 pygpilog.exp
-a----        30/10/2024     11:17           2272 pygpilog.lib
```

Is NVC support not enabled on Windows builds ?

Environment : 
- Cocotb 1.9.2 and latest master
- Windows 64 bits
- NVC simulator 1.14.1
- Python 3.13

themperek, 2024-11-02:
It looks like NVC is not supported on Windows. One would need to update `cocotb_build_libs.py`

PteronuraBrasiliensis, 2025-11-10:
I have the same issue on Windows with cocotb 2.0.0

`** Note: loading VHPI plugin
         C:/Users/.../AppData/Local/Programs/Python/Python311/Lib/site-packages/cocotb/libs/cocotbvhpi_nvc.dll`
`** Fatal: failed to load
          C:\Users\...\AppData\Local\Programs\Python\Python311\Lib\site-packages\cocotb\libs\cocotbvhpi_nvc.dll:
          The specified module could not be found.`

Since it seems to be related to a problem within `cocotb_build_libs.py`, why is it labeled as type:feature and not type:bug ?

ktbarrett, 2025-11-10:
NVC is not supported on Windows in 2.0. This support was just added to master and was also backported to be a part of the 2.0.1 release. Considering this has be merged I'll close the issue. Look for 2.0.1 out in the coming days or weeks.

closed 2025-11-10 by ktbarrett