cocotb/cocotb #4250: DLL not found when using Questa Visualizer

MAtten-Ametek, 2024-11-07:
The 'adder' example works for me with Questa using the classic (ModelSim-style) GUI, but not with the Visualizer UI.  Passing `gui=True` and including `'-visualizer'` in the test_args list for runner.test seems to be enough to trigger the problem.
The symptoms are that after loading cocotbvpi_modelsim.dll, the log/transcript shows an error saying that the python312.dll could not be opened.  I checked and the given path for the DLL is for a file that does exist,
```
# Loading sv_std.std
# Loading work.adder(fast)
# Loading C:/Users/Michael.Attenborough/AppData/Local/miniconda3/Lib/site-packages/cocotb/libs/cocotbvpi_modelsim.dll
     -.--ns ERROR    gpi                                ..s\cocotb_utils.cpp:56   in utils_dyn_open                  Unable to open lib C:\Users\Michael.Attenborough\AppData\Local\miniconda3\python312.dll: The specified module could not be found.

     -.--ns INFO     gpi                                ..\gpi\GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     gpi                                ..\gpi\GpiCommon.cpp:101  in gpi_print_registered_impl       FLI registered
```
```
>dir C:\Users\Michael.Attenborough\AppData\Local\miniconda3\python312.dll
 Volume in drive C is Windows
 Volume Serial Number is xxxx-xxxx

 Directory of C:\Users\Michael.Attenborough\AppData\Local\miniconda3

04/10/2024  13:27         7,457,040 python312.dll
               1 File(s)      7,457,040 bytes
```
The rest of the Verilog design loads, but not surprisingly there is no Python integration.
My environment:
- cocotb 1.9.2
- python 3.12.7 (miniconda)
- QuestaSim Base Edition-64 version 2024.3
- Windows 10 Version 2009, kernel 10.0.19045 x86_64

To simplify the reproduction of the issue, I made some modifications to test_adder.py so that it takes an argument to control whether the test is run in text mode (`python test_adder.py text`), the classic GUI (`gui`) or Visualizer (`vis`).   My modified version is [here](https://github.com/MAtten-Ametek/cocotb/blob/visualizer_dll_bug_demo/examples/adder/tests/test_adder.py).  In the case of the Visualizer mode, it also adds a `-qwavedb=+signal+memory+vhdlvariable` switch which I have been recommended to use, but leaving this out makes no difference to the results.
I have attached log files from Visualizer ([visualizer.log](https://github.com/user-attachments/files/17657738/visualizer.log)) and also a log from the classic GUI ([transcript](https://github.com/user-attachments/files/17657747/transcript.txt)) for comparison.
I asked my local support engineer to look at this, and interestingly the problem does not occur on his system, which suggests to me that the problem is not unsurmountable.  He tried it on Windows with cocotb 1.9.1, python 3.12.2 and Questa 2024.3.

MAtten-Ametek, 2024-11-07:
I got the python 3.12.2 installer from my Questa support engineer, which is a "python for Windows" installer, not miniconda.  After installing cocotb 1.9.2 in this python, the problem does not occur.
I then got the python-3.12.7-amd64.exe installer from https://www.python.org/downloads/windows/, installed cocotb 1.9.2 in that, and the problem does not occur there either.
So on the face of it, the problem is linked to the use of miniconda instead of the "python for Windows" python installation.
At least I have a workaround, which is good, but the cocotb documentation recommends the use of miniconda for native Windows development so maybe this issue should remain open.

themperek, 2025-01-02:
Likely do to path style Windows, tcl, and python. It is close to impossible to get this correct on Windows.