cocotb/cocotb #4259: How to get the current timing model stage from python code?

myrfy001, 2024-11-09:
Ihave read the timing model document from wiki page, and I have some questions.

Suppose I want to write a function that will block on a ready signal, the basic idea is like that:
```
async def wait_ready_and_do_something():
        while not ready_signal.value:
            await RisingEdge(ready_signal)
        # Do something useful things here
```

But the above code has problems: we don't know in which timing model state that this function is invoked.
for example, the user may use this function in the following two ways:
```
async def foo():
    await RisingEdge(clk)
    # we are in  `Beginning of Time Step` state described in the timing model
    await wait_ready_and_do_something()
```
OR
```
async def foo():
    await RisingEdge(clk)
    await some_user_function()
    # now, we are not sure what timing model state we are in, the above `some_user_function` may contain code
    # like await ReadWrite(), we can't control that.
    await wait_ready_and_do_something()
```

if the code is like the first usage example, then we are in the `Beginning of Time Step` state, the ready signal may not be the newest value, so the `wait_ready_and_do_something()` function will have wrong beahvior.

one of the way to fix the bug is to move the `await ReadWrite()` into the `wait_ready_and_do_something()` function like that:
```
async def wait_ready_and do_something():
        await ReadWrite()
        while not ready_signal.value:
            await RisingEdge(ready_signal)
        # Do something useful things here
```

But if the above modified version function is called in a `Value Change` timing model state, from the wiki, it said that 

> Waiting on ReadWrite again after a ReadWrite will result in undefined behavior and may one day result in a Python exception.

I think the best way to solve this problem is to add a guard to our function like that:
```
async def wait_ready_and do_something():
        if cocotb.curent_timing_model_state == cocotb.BEGIN_OF_TIME_STEP:
                await ReadWrite()
        while not ready_signal.value:
            await RisingEdge(ready_signal)
        # Do something useful things here
```

so, is there any way to get the current state in which the python code is executed?

Thanks.

myrfy001, 2024-11-09:
After digging into the code, I have found that the master branch now has a `cocotb.sim_phase` that can get the sim phase, which I think is just what I need, but I still have two questions:

1. it seems not to be included to released version until now (1.9.2), so in which version will it have this released?
2. it seems that there are just 3 phase: `Normal`, `ReadWrite`, `ReadOnly`, which doesn't matched the timing model wiki page's 4 state. So any detail about that?

Thanks.

ktbarrett, 2024-11-13:
> 1. it seems not to be included to released version until now (1.9.2), so in which version will it have this released?

It will be released in 2.0 which should be complete around the end of this year or early next year.

> 2. it seems that there are just 3 phase: `Normal`, `ReadWrite`, `ReadOnly`, which doesn't matched the timing model wiki page's 4 state. So any detail about that?

Originally this variable was a variable in the scheduler and it only tracked "Normal" and `ReadOnly`. We can certainly expand this to differentiate Value Change from BOTS.

closed 2024-11-14 by myrfy001