cocotb/cocotb #4261: analog/mixed signal simulation using cocotb and vcs

HRET6288, 2024-11-11:
Hi, 

I'm trying to run an analog/mixed signal simulation using cocotb and vcs ( i.e the top module is in verilog, and there is an analog subblock that is simulated using the spice engine (primesim SPICE in my case)). 
When no SPICE block is included, the outputs are fine. When one block ( a simple buffer in my case) is turned into SPICE, I see the following funny issues: 
1- The supply pins of the spice subblock are propagated to the top level. For instance, if the SPICE subblock is 3 levels down from top, the vdd and vss pin of the top verilog block ( which should be digital and is coming from my python script) are changed from a digital value to an analog voltage.
2- The top level vdd and vss values ( which should be 1 and 0 respectively) are toggling as the input pin of the SPICE block.

The warnings that I get are:  
******************************************************************************
__Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.
  Please fix the code being referred to by the '-load' compile option.

     -.--ns ERROR    gpi                                ..b/vpi/VpiCbHdl.cpp:72   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbStartOfSimulation(11)

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.
  Please fix the code being referred to by the '-load' compile option.

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_register_cb' is not supported in the compiler.
  Please fix the code being referred to by the '-load' compile option.

     -.--ns ERROR    gpi                                ..b/vpi/VpiCbHdl.cpp:72   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbEndOfSimulation(12)

Warning-[VPI-CT-NS] VPI function is not supported
  The VPI function 'vpi_chk_error' is not supported in the compiler.
  Please fix the code being referred to by the '-load' compile option.__
*****************************************************************************

The tool versions: 
Python 3.9.12
Cocotb-config --version: 1.9.1
VCS: T-2022.06-SP2
Primesim SPICE: T-2022.06-SP2

Does anyone know how I can resolve this? 

Thanks
HRET

ktbarrett, 2024-11-14:
VCS has a number of issues running cocotb, so I can't say I'm surprised you are seeing these errors. Have you tried a newer version of VCS? I expect it to have better VPI support.

themperek, 2025-01-02:
Those warnings very likely do not matter and are some other issues.
 
You can try not to drive any analogue pin (vdd,vss, ..) from the top of the testbench (python). Use a wrapper if needed?