cocotb/cocotb #4269: Unable to build multiple toplevel hdl files using iverilog

aamodini31, 2024-11-14:
I have two hdl modules - counter_4.v and counter_24.v (4-bit and 24-bit counters respectively) and their corresponding cooctb test cases are written in tb_counter4b.py and tb_counter24b.py. I am trying to build and run both these test files one after the other using py runner script but I get the following error:

0.00ns INFO     cocotb                             Seeding Python random module with 1731622593
     0.00ns ERROR    gpi                                VPI: Couldn't find root handle Counter24Bit
     0.00ns ERROR    gpi                                VPI: Toplevel instances: Counter24Bit != Counter4Bit...
     0.00ns ERROR    gpi                                No root handle found

Runner script as follows:
[image]

Is there an alternate way to build both the hdl top modules together and run using the iverilog simulator?

teobiton, 2024-11-15:
Hi @aamodini31, you should add `always = True` to your build function. Right now, when reaching the second test, the RTL is not rebuilt, so cocotb tries to run your second test on the same module. By setting the `always`  flag, the module will be rebuilt with the correct sources in the second test.

aamodini31, 2025-01-08:
Got it, thank you!

closed 2025-01-08 by aamodini31