cocotb/cocotb #4271: GHDL: fatal error on accessing value property of array of SLV signal constrained by generic

JvWesterveld, 2024-11-18:
Accessing the `value` property of the `mem` signal in the below VHDL snippet throws a fatal error.
```
library ieee;
  use ieee.std_logic_1164.all;

entity dut is
  generic(
    DATA_WIDTH : positive := 8
  );
end entity;

architecture rtl of dut is
  type t_mem is array (natural range 0 to 1) of std_logic_vector(DATA_WIDTH - 1 downto 0);
  signal mem : t_mem;
begin
end architecture;
```

```
import cocotb

@cocotb.test()
async def some_test(dut):
  print(dut.mem.value)
```

```
user@host:~/reproducer$ nix-shell

[nix-shell:~/reproducer]$ make
rm -f results.xml
make -f Makefile results.xml
make[1]: Entering directory '/home/user/reproducer'
mkdir -p sim_build
\
/nix/store/ym0zzc1ixlgraf114nabkyr664r41537-ghdl-mcode-4.1.0/bin/ghdl -i   --workdir=sim_build --work=work dut.vhdl && \
/nix/store/ym0zzc1ixlgraf114nabkyr664r41537-ghdl-mcode-4.1.0/bin/ghdl -m   --workdir=sim_build -Psim_build --work=work dut
rm -f results.xml
MODULE=tb TESTCASE= TOPLEVEL=dut TOPLEVEL_LANG=vhdl \
 /nix/store/ym0zzc1ixlgraf114nabkyr664r41537-ghdl-mcode-4.1.0/bin/ghdl -r  --workdir=sim_build -Psim_build --work=work dut --vpi=/nix/store/mrmppzlbwra265vd9q3fki2vcmnzld9f-python3.12-cocotb-1.8.1/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_ghdl.so  
loading VPI module '/nix/store/mrmppzlbwra265vd9q3fki2vcmnzld9f-python3.12-cocotb-1.8.1/lib/python3.12/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
VPI module loaded!
     0.00ns INFO     cocotb                             Running on GHDL version 4.1.0 (tarball) [Dunoon edition]
     0.00ns INFO     cocotb                             Running tests with cocotb v1.8.1 from /nix/store/mrmppzlbwra265vd9q3fki2vcmnzld9f-python3.12-cocotb-1.8.1/lib/python3.12/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1731837024
     0.00ns INFO     cocotb.regression                  pytest not found, install it to enable better AssertionError messages
     0.00ns INFO     cocotb.regression                  Found test tb.some_test
     0.00ns INFO     cocotb.regression                  running some_test (1/1)
ghdl:error: NULL access dereferenced
     0.00ns ERROR    gpi                                FATAL: We are calling up again
make[1]: *** [/nix/store/mrmppzlbwra265vd9q3fki2vcmnzld9f-python3.12-cocotb-1.8.1/lib/python3.12/site-packages/cocotb/share/makefiles/simulators/Makefile.ghdl:77: results.xml] Error 1
make[1]: Leaving directory '/home/user/reproducer'
make: *** [/nix/store/mrmppzlbwra265vd9q3fki2vcmnzld9f-python3.12-cocotb-1.8.1/lib/python3.12/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

Environment:

- GHDL 4.1.0 64-bit (tested mcode and GCC backends)
- Cocotb 1.8.1
- Ubuntu 24.04.1 LTS x64
- Using nix-shell to set up environment

I put all files that I used in the reproducer in [this gist](https://gist.github.com/JvWesterveld/6da692fc54bbb82df600b789ed5f3095).

Is this the right place for this issue report or should it be with the GHDL project?

cmarqu, 2024-11-18:
Let's ping @tgingold and see if that report is enough.

tgingold, 2024-11-23:
Thank you for the testcase, it is now fixed in ghdl.

SeanBradley2625, 2024-12-03:
I though you couldn't access arrays greater than one dimension over VPI? I could be wrong and would happily be corrected as I have been trying to get this to work with GHDL.

ktbarrett, 2024-12-03:
@SeanBradley2625 You can't indexing into *packed* arrays at all with the VPI. Is that what you're thinking of? But indexing into N dimensional unpacked arrays via the VPI should be possible. 

One exception is the fact that GHDL uses the VPI and arrays like `std_logic_vector` or `unsigned` are treated as "packed" though they may not otherwise be.

SeanBradley2625, 2024-12-03:
@ktbarrett Would the above example not be considered a packed array?

ktbarrett, 2024-12-05:
@SeanBradley2625 `mem[0]` should succeed. `mem[0][0]` will not.

JvWesterveld, 2024-12-16:
This is indeed fixed with https://github.com/ghdl/ghdl/commit/39c0b9c79cc2fbadfa42faccc0fc73cd3321d545. Thanks!

closed 2024-12-16 by JvWesterveld