cocotb/cocotb #4274: subscriptable signals (LogicObject)

psychogenic, 2024-11-20:
Greetings,

I'm wondering if I'm missing something obvious.  Exploring cocotb v2.0, I wanted to access and modify individual bits and slices from a signal.

I wound up implementing https://github.com/psychogenic/cocotb/commit/0486f5077aa4914c35af6bbeb5ef855c21c686b3

Assuming an interface like the adder
```
module adder #(
  parameter integer DATA_WIDTH = 4
) (
  input  logic unsigned [DATA_WIDTH-1:0] A,
  ...
);
```
This allows for what I would think is a "natural" interface to the individual portions of the signal, for reads and writes, e.g.
```
   # the usual
   dut.A.value = 0
   # but also 
   dut.A[0] = 1
   # or slice assignments
   dut.A[3:2] = 0b11
```
Doing such operations on the _A.value_ is possible, but since the _LogicObject_ returns a freshly instantiated _LogicArray_ on every call to value, and this object is oblivious to the parent it was spawned from, any such changes are lost to the ether.  

My questions are:

- is there some obvious method for doing this that I just missed?
- if not, would making test cases for this patch be worth doing in order to create a PR?
- if the PR is desired, what changes to the interface or functionality would make it betterer?

Thanks!

psychogenic, 2024-11-24:
**Single-bit/Slice manipulations**
Ok, so my original patch above uses the regular `port.value` for the whole byte and `port[]` for individual bits/slices.  It works but has some weirdness to it (namely because reading `port.value[idx]` works but writing to it, `port.value[idx] = 1`, won't) so you had to do `port[idx] = 1`.

I've now gone in and figured an easy way to get the returned _LogicArray_ a bit more aware so that value indexing actually work how you would expect, i.e. reads work but also
```
dut.A.value[0] = 1
dut.A.value[3:1] = 0b11 # or [1, 1] or any iterable of the right size
```
You can see these minor changes in https://github.com/cocotb/cocotb/commit/a86e8fd2f7b6ae13c1219169fa9973a42c59cd42

marlonjames, 2024-12-11:
This is usually accomplished with a read-modify-write sequence, such as:

```py3
temp = dut.A.value
temp[3:1] = 3
dut.A.value = temp
```
If you want the convenience of a one-liner, you could also write your own wrapper class that takes the `A` handle and implements `__setitem__` slicing that can do the RMW internally.

ktbarrett, 2024-12-12:
Following on what @marlonjames said:

```python
class update:

    def __init__(self, signal: LogicObject) -> None:
        self.signal = signal

    def __setitem__(self, index: int | slice, value: int | str | Sequence[int | str | Logic]) -> None:
        temp = handle.value
        temp[index] = value
        handle.value = temp
```

Usage:
```python
update(dut.A)[0] = 1
update(dut.A)[3:2] = 0b11
```

closed 2024-12-12 by ktbarrett