cocotb/cocotb #4284: Duplicate addition of cocotb_iverilog_dump.v in VERILOG_SOURCES when using make -f

yizishun, 2024-12-01:
make version = 3.81

When using `make -f` to invoke the Makefile.sim, the file `cocotb_iverilog_dump.v` 
is appended multiple times to the `VERILOG_SOURCES` variable. This causes duplicate module 
declarations and results in compilation errors during simulation.

**Reproduce**
I modify [quckstart](https://github.com/cocotb/cocotb/tree/master/examples/doc_examples/quickstart) Makefile as follows:

```makefile
# This file is public domain, it can be freely copied without restrictions.
# SPDX-License-Identifier: CC0-1.0

# Makefile

# defaults
SIM ?= icarus
TOPLEVEL_LANG ?= verilog
WAVES = 1

VERILOG_SOURCES += $(PWD)/my_design.sv
# use VHDL_SOURCES for VHDL files

# TOPLEVEL is the name of the toplevel module in your Verilog or VHDL file
TOPLEVEL = my_design
WAVES = 1

# MODULE is the basename of the Python test file
MODULE = test_my_design

COCOTB_MAKEFILE = $(shell poetry run cocotb-config --makefiles)/Makefile.sim
export
# include cocotb's make rules to take care of the simulator setup
sim:
	poetry run \
	make \
	-f $(COCOTB_MAKEFILE)

```

and run `make sim` and get
```
poetry run \
	make \
	-f /Users/yizishun/Library/Caches/pypoetry/virtualenvs/cocotb-try-fuIVHJhZ-py3.12/lib/python3.12/site-packages/cocotb/share/makefiles/Makefile.sim
rm -f results.xml
"/Library/Developer/CommandLineTools/usr/bin/make" -f /Users/yizishun/Library/Caches/pypoetry/virtualenvs/cocotb-try-fuIVHJhZ-py3.12/lib/python3.12/site-packages/cocotb/share/makefiles/Makefile.sim results.xml
mkdir -p sim_build
/etc/profiles/per-user/yizishun/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s my_design -g2012 -f sim_build/cmds.f -s cocotb_iverilog_dump  /Users/yizishun/test/cocotb_try/my_design.sv sim_build/cocotb_iverilog_dump.v sim_build/cocotb_iverilog_dump.v
sim_build/cocotb_iverilog_dump.v:1: error: 'cocotb_iverilog_dump' has already been declared in this scope.
sim_build/cocotb_iverilog_dump.v:1:      : It was declared here as a module.
sim_build/cocotb_iverilog_dump.v:6: error: Module cocotb_iverilog_dump was already declared here: sim_build/cocotb_iverilog_dump.v:1

make[2]: *** [sim_build/sim.vvp] Error 2
make[1]: *** [sim] Error 2
make: *** [sim] Error 2
```

ktbarrett, 2024-12-02:
Do you understand why it's being appending twice? I only see how the Makefile included/run once.

yizishun, 2024-12-03:
i think the main reason is the sim target in Makefile.inc, which is Unconditionally re-run the simulator (default). And i export the VERILOG_SOURCES, so it is appended twice.

yizishun, 2024-12-03:
I just add an `export` in [quckstart](https://github.com/cocotb/cocotb/blob/master/examples/doc_examples/quickstart/Makefile) example and also reproduce this problem.So I want to add `VERILOG_SOURCES := $(filter-out $(SIM_BUILD)/cocotb_iverilog_dump.v, $(VERILOG_SOURCES))` before appending cocotb_iverilog_dump.v to solve this

ktbarrett, 2024-12-03:
I feel like there is probably a smarter way. I'll give it some thought soon. Do you have a work around for now?

yizishun, 2024-12-04:
Thank you in advance for your patience. But now the approach above is the only solution I counld find.